FreeCAD/FreeCAD #18006: Sketcher lines lacking further width customization

SpeeQz1, 2024-11-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently the only way to modify the thickness of certain sketcher lines is through `View scale ratio`, the problem with that is that it edits the thickness of every line rather than specific ones.

A proposition is to add options for individually editing the line thickness of all type of lines such as dimensional constraints for example.

[image]

### Full version info

```shell
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * fasteners 0.5.29 (Disabled)
  * lattice2 1.0.0
  * Manipulator 1.5.7
  * CurvedShapes 1.0.13
  * FreeCAD-Ribbon 1.4.0
  * SearchBar 1.0.2
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

obelisk79, 2024-11-20:
I believe it would be useful. At least as hidden parameters that can be assigned by themes and advanced users.

obelisk79, 2024-11-20:
You can do this only for some elements

paullee0, 2025-01-20:
Good to have python extension to support this feature also (sketcher::pythonFeature ?)