FreeCAD/FreeCAD #18010: macOS x86 build doesn't respect dark mode (arm version does)

levito, 2024-11-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The macOS x86 build of FreeCAD 1.0 final doesn't respect dark mode: There is a bright title bar and open/save dialogs, and some UI elements are hardly readable.

The macOS arm build does respect dark mode, and the x86 build of RC2 did as well.

**To Reproduce**
Steps to reproduce the behavior:

On an Apple silicon Mac, download both the arm and x86 builds of FreeCAD 1.0
Switch to dark mode in macOS preferences
Open the arm build: the title bar will be dark, as it should
Open the x86 build: the title bar will be bright, but it should be dark as well

**Screenshots**
See the arm build on the left, the x86 build on the right:

[image]

### Full version info

```shell
OS: macOS 14.7
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * OpenTheme 2024.9.1
  * Marz.backup1731717103.49366 0.1.11 (Disabled)
  * MnesarcoUtils 0.2.14
  * Marz 0.1.12
```

### Subproject(s) affected?

None

### Anything else?

I'm not sure if this is an issue in the software itself or related to bundling. Therefore, I also created another issue here: https://github.com/FreeCAD/FreeCAD-Bundle/issues/340

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-20:
@MisterMakerNL FYI

hyarion, 2024-11-20:
@levito It might be that there's an earlier config on the system, can you try to run it in safe mode on the one that doesn't respect dark mode? (menu > help > restart in safe mode)

levito, 2024-11-20:
@hyarion Both are running from the same config, even at the same time. I took the screenshots showing the arm and x86 version on an M3 Mac and I'm also running an x86 installation on an Intel Mac, which shows the same behavior. Also tried deleting the config and starting fresh.

On the Intel Mac, the RC2 also respects the dark mode (though it has other visual issues like unreadable text, i. e. white on white or black on dark)

The arm version with Classic theme also shows white text on light gray background (see 2nd screenshot left), but that's not what this issue is about.

hyarion, 2024-11-20:
Anyone else with an x86 mac who could verify this issue?

levito, 2024-11-21:
You can also verify it with an Apple silicon Mac, just run the x86 build via Rosetta.

hyarion, 2024-11-21:
I tried it using rosetta, and indeed when the os is in dark mode, the qt title bars are white etc.
The auto-select light/dark mode feature correctly detects the system mode and selects the theme accordingly.

Do we do anything special to allow light/dark mode on arm64 that we don't do on x86? Or is it a Qt 5 bug?

antgeth, 2024-11-23:
can confirm the issue on an x86 mac

luzpaz, 2024-11-23:
@antgeth thanks, though please always paste your full About info when you confirm something, TIA!

maxwxyz, 2026-03-08:
I cannot reproduce anymore

closed 2026-03-08 by maxwxyz