FreeCAD/FreeCAD #18011: BIM: Error when switching to strict IFC mode in 1.1dev

semhustej, 2024-11-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

- Open a new file
- Switch project to Strict IFC mode
- Error shows up

```
18:43:06  Traceback (most recent call last):
18:43:06    File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39240-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_status.py", line 70, in on_toggle_lock
18:43:06      lock_document()
18:43:06    File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39240-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_status.py", line 188, in lock_document
18:43:06      from nativeifc import ifc_tools  # lazy loading
18:43:06      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
18:43:06    File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39240-conda-Windows-x86_64-py311\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
18:43:06      return original_import(name, *args, **kwargs)
18:43:06             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
18:43:06    File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39240-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_tools.py", line 36, in <module>
18:43:06      from ifcopenshell import geom
18:43:06    File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39240-conda-Windows-x86_64-py311\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
18:43:06      return original_import(name, *args, **kwargs)
18:43:06             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
18:43:06    File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39240-conda-Windows-x86_64-py311\bin\Lib\site-packages\ifcopenshell\geom\__init__.py", line 55, in <module>
18:43:06      from . import occ_utils as utils
18:43:06    File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39240-conda-Windows-x86_64-py311\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
18:43:06      return original_import(name, *args, **kwargs)
18:43:06             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
18:43:06    File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39240-conda-Windows-x86_64-py311\bin\Lib\site-packages\ifcopenshell\geom\occ_utils.py", line 27, in <module>
18:43:06      from typing_extensions import assert_never
18:43:06    File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39240-conda-Windows-x86_64-py311\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
18:43:06      return original_import(name, *args, **kwargs)
18:43:06             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
18:43:06  ModuleNotFoundError: No module named 'typing_extensions'

```

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39240 (Git) Conda
Build type: Release
Branch: main
Hash: f9ba94c70b139ece6ba5ffba47632d91b5c9c645
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US) [ OS: English/United States (en_US) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

BIM

### Anything else?

I am not able to reproduce the issue with FC 1.0, it seems like a regression.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-12-05:
Hmm `typing_extensions` is a non-builtin python module, and it seems to be required by ifcopenshell.

@Andrej730 can you confirm? We might need to start adding it to the FreeCAD bundle...

Andrej730, 2024-12-05:
@yorikvanhavre Hi! Yes, I can confirm, `typing-extensions` is required since we use some typing features from newer Python versions and we still support Python 3.9-3.10.

yorikvanhavre, 2024-12-05:
Thanks @Andrej730 !

@adrianinsaval is it doable to include `python3-typing-extensions` in our packages?

maxwxyz, 2025-01-07:
Is this still an issue in latest weekly with https://github.com/FreeCAD/FreeCAD/pull/18769 ?

semhustej, 2025-01-07:
I did a test and the issue seems to be fixed on my side with the latest weekly. When reproducing, I only get this information (no errors):

```
12:23:12  Note: API not available due to missing dependencies: geometry.add_door_representation - No module named 'mathutils'
12:23:12  Note: API not available due to missing dependencies: geometry.add_railing_representation - No module named 'mathutils'
12:23:12  Note: API not available due to missing dependencies: geometry.add_representation - No module named 'bpy'
12:23:12  Note: API not available due to missing dependencies: geometry.add_window_representation - No module named 'mathutils'
12:23:12  Note: API not available due to missing dependencies: grid.create_axis_curve - No module named 'mathutils'
12:23:12  DEBUG: Extra recomputing of 0 objects took 00:00
12:23:12  DEBUG: Extra recomputing of 0 objects took 00:00
```

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39750 (Git) Conda
Build type: Release
Branch: main
Hash: 7b548bee7f781837640df96831a3e7a7cfa4cf9b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

```

yorikvanhavre, 2025-01-07:
These "Note: API not available due to missing dependencies" messages are from the latest ifcopenshell version... Apparently it has some stuff that depends on Blender (mathutils is a [blender module](https://docs.blender.org/api/current/mathutils.html))

But it does not seem to have any effect in FreeCAD, everything still works normally.

maxwxyz, 2025-01-07:
Closing then as fixed by https://github.com/FreeCAD/FreeCAD/pull/18769

closed 2025-01-07 by maxwxyz