FreeCAD/FreeCAD #18015: Blender Navigation Panning with left/right mouse buttons

steveman1123, 2024-11-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

### Action:
while using Blender navigation mode:
press and hold left mouse button, and press and hold right mouse button

### Expected Behavior:
switch from selection mode to panning mode

### Actual Behavior:
selection mode stays active. In order to use double mouse button panning mode in Blender navigation, right button then left button must be pressed (order dependent).

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.38641 +468 (Git) Snap 1199
Build type: Release
Branch: tag: 1.0.0
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * A2plus 0.4.61
  * CurvedShapes 1.0.5
  * Manipulator 1.5.6
  * Curves 0.6.22
```

### Subproject(s) affected?

None

### Anything else?

Low priority, workaround to press right MB followed by Left MB

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-11-20:
cc @Rexbas

Rexbas, 2024-11-20:
If I press LMB first and then RMB it switches to panning mode which is expected right?!. Not sure I understand the problem... @steveman1123 could you share a video?

steveman1123, 2024-11-20:
I just went and tried again, and it does switch as expected when in the general 3D view (not sure what it's called exactly), but then it does perform the unexpected behavior when editing a sketch. [Link to video](https://www.technicallysane.com/stevenw/links/freecad-issue.mp4) - apologies for the poor quality, haven't had a chance to get OBS set up well

Rexbas, 2024-11-20:
Ah yes. In editing mode (sketch), some actions are disabled to prevent navigation conflicts with the rubber band selection tool. I do think that pressing LMB and then immediately RMB without moving the mouse should activate panning mode and cancel rubber band selection. However, it is not an easy fix right now and needs refactoring. I will add this to my navigation refactoring TODO list.

steveman1123, 2024-11-20:
I can't seem to get that to work, (left then immediate right), but learning to do right then left seems to do fine too. Thanks for the quick responses! Not a high priority

Rexbas, 2024-11-20:
> I can't seem to get that to work, (left then immediate right)

I mean, that is how it should work when it is fixed, right now that does not work.

xobes, 2025-01-20:
Is there not a way to customize each key/button-press combo that will initiate pan/rotate/zoom?  I don't get why we have to choose from pre-canned stuff.  Cura has a really intuitive setup that doesn't require two hands.  Plus I have a touchscreen, so some aspects of that would be nice to support in my setup as well -- but there is only pre-canned setups.

> If I press LMB first and then RMB it switches to panning mode which is expected right?!. Not sure I understand the problem... [@steveman1123](https://github.com/steveman1123) could you share a video?

This works for me, and I had no clue about it, so at least I learned something today :-)

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Assembly4 0.50.15
  * Curves 0.6.53
  * freecad.gears 1.3.0
  * InventorLoader 1.5.1
  * sheetmetal 0.6.13
  * slic3r-tools

xobes, 2025-01-25:
> This works for me, and I had no clue about it, so at least I learned something today :-)

... In the Part workbench.  Not in Sketcher.

PR #22998 (2025-09-01): Navigation: Prevent showing context menu after panning or rubber band selection
Fixes #22931 and similar problems in other navigation styles.
Fixes #18015.

Postpones resetting `hasPanned`, `hasDragged` and `hasZoomed` until the navigation mode changes from IDLE to something else or if all mouse buttons have been released. This preserves information about the last navigation action while in IDLE mode until all mouse buttons have been released. This is needed for the blender style because when panning with LMB and RMB if LMB is released the mode goes to IDLE. But to prevent showing the context  [... cut]