FreeCAD/FreeCAD #18019: Scaling issue with "Close Without Saving" button in unsaved document dialog

Condyloma, 2024-11-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When working on a project in FreeCAD that hasn’t been saved and attempting to close the program using the "X" button in the top-right corner, the warning dialog about unsaved changes appears as expected. However, there is a scaling issue with the "Close Without Saving" button. The button does not display fully, and part of it is cut off.

### Full version info

```shell
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

huguesdpdn-aerospace, 2024-11-20:
For my first time, I tried to fix this by myself.

## Investigations start
I found the C++ code here: https://github.com/FreeCAD/FreeCAD/blob/2819ca438e3776fdb675fbe061b95019b70940d4/src/Gui/MainWindow.cpp#L867-L922

## FreeCAD or Qt ?
However, as far as I can see, this is not handled in FreeCAD. I mean, FreeCAD do not  add/place the buttons and the icons, neither control predefined buttons, neither create the message box (because it is actually a `QMessageBox`)
FreeCAD does not even control the translation (actually translation is handled here: https://github.com/qt/qttranslations/blob/dev/translations/qtbase_fr.ts#L3783 )

## Themes checks
As @Condyloma , I have `FreeCAD Dark.qss/FreeCAD Dark/Fusion` theme. :
https://github.com/FreeCAD/FreeCAD/blob/2819ca438e3776fdb675fbe061b95019b70940d4/src/Gui/PreferencePacks/FreeCAD%20Dark/FreeCAD%20Dark.cfg#L177-L183
Also, I look into file `FreeCAD Dark.cfg` 
https://github.com/FreeCAD/FreeCAD/blob/2819ca438e3776fdb675fbe061b95019b70940d4/src/Gui/Stylesheets/FreeCAD%20Dark.qss#L1618-L1623
But not sure the first comment line is linked to our issue.

## Requesting FreeCAD senior DEV help

So I still don't know if it is a FreeCAD theme issue or a QtStyle `Fusion` issue.

...

Help :)

yomgui1, 2025-03-16:
Hi.
The issue here is caused by setting the **min-width** property of QPushButton into the theme stylesheet file.
For example here for the dark theme:

https://github.com/FreeCAD/FreeCAD/blob/9a6a98f824436656cae242ae26b043f78411ebba/src/Gui/Stylesheets/FreeCAD%20Dark.qss#L1296C1-L1298C2

In the dark theme, it's set to 20 pixel. So when you get a long text file (like the one for the checkbox added on the left), all buttons are shrink until they reach this minimal size.

As Qt documentation mention about the minimalSizeHint:

> [QLayout](https://doc.qt.io/qt-5/qlayout.html) will never resize a widget to a size smaller than the minimum size hint unless [minimumSize](https://doc.qt.io/qt-5/qwidget.html#minimumSize-prop)() is set or the size policy is set to QSizePolicy::Ignore. If [minimumSize](https://doc.qt.io/qt-5/qwidget.html#minimumSize-prop)() is set, the minimum size hint will be ignored.

So the solution is to set this minimal size to -1.

Here a capture of where I've hacked the code to add a very long checkbox text and using a French UI to have a long Cancel button text. I've set to -1 the min-width for the confirmSave window:

[image]

The same with the orignal "20 px" value:

[image]

yomgui1, 2025-03-17:
BTW, general note, using a minimal fixed size (like 20px or even the 80px I can see just below in the qss) for button with translated text is not a good idea as we can see here. Except of very good reason, we should let the layout engine computes that for us.

PR #20241 (2025-03-19): fix dialogs button size in case of long text (fix for #18019)
Let the layout engine computes the size of all QPushButton by unset the minimalSize value set by "min-width" QSS property in themes.

Fix #18019.

**Before**

[image]

**After**

[image]
commit e127f2c8 (2025-03-19): fix dialogs button size in case of long text

Let the layout engine computes the size of all QPushButton by unset
the minimalSize value set by "min-width" QSS property in themes.

Fix #18019.
closed 2025-03-19 by kadet1090