FreeCAD/FreeCAD #18024: Model disappears on transform -> position becomes NaN -> seems related to python?

JustTryingToGetSomeWorkDone, 2024-11-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I translate a simple group and it typically will work once, but if you stop and then attempt to translate in a second direction it will almost always cause the model to disappear. If you rotate the view every other model will also disappear. The position of the translated group becomes "nan m". The position cannot be manually changed due to a problem that seems to be related to python. Clicking center on model doesn't help, Undoing will typically make it reappear in recent versions. This has been present in every version of FreeCAD I've used, but it has gotten better in more recent versions, but clearly it's still happening.

After attempting to change the position the following error displays in the message area. This is the only error that is ever displayed:

pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'NameError'>: name 'nan' is not defined

The attached video files are actually mkv files renamed to be uploadable. The underlying video stream is indeed mp4 compatible. They should play in VLC or anything that uses ffmpeg. The first shows the model disappearing, then towards the end the program hangs. I stopped the video recording. Then later it began responding again so I started another recording and show the "nan m" positions and the error message and finally undoing to get the models back.

https://github.com/user-attachments/assets/21aba653-173a-430f-87be-947f1922d0dd

https://github.com/user-attachments/assets/63fcebd5-ede3-492a-89ec-eb39adb5c79b

This is a work file, so I can upload privately, but do not want to post it publicly. Any sufficiently complex parts added to a simple group should show the same behavior I would assume.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US) [ OS: Japanese/Japan (ja_JP) ]
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Classic/Qt default
Installed mods: 
  * Curves 0.6.47
  * fasteners 0.5.28
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

JustTryingToGetSomeWorkDone, 2024-11-21:
https://github.com/user-attachments/assets/dee84b06-f49e-44b1-8794-8ab08adc7ede

https://github.com/user-attachments/assets/e018af7b-8e02-4cfb-89a7-b0635283ce5f

maxwxyz, 2024-11-21:
@JustTryingToGetSomeWorkDone can you create a simple demo file to be able to reproduce the issue?
@kadet1090 is this something you've come across during your transform overhaul?

JustTryingToGetSomeWorkDone, 2024-11-21:
> @JustTryingToGetSomeWorkDone can you create a simple demo file to be able to reproduce the issue? @kadet1090 is this something you've come across during your transform overhaul?

I believe so. It has happened to me many times, so it should be easy enough to make an unimportant file where it happens.  I'll see what I can get done in the next 20 minutes. It's getting late here, so it might not be until tomorrow that I get it breaking reliably.

maxwxyz, 2024-11-21:
you can upload a .zip

JustTryingToGetSomeWorkDone, 2024-11-25:
[recreateFreeCADbug_060415.zip](https://github.com/user-attachments/files/17895736/recreateFreeCADbug_060415.zip)

Sorry, looks like my company's firewall was preventing my upload. Here is the prepared demo file.

JustTryingToGetSomeWorkDone, 2024-11-25:
Here's a video of basically the same thing happening, but this time the position doesn't seem to go to NaN, and there is never an error in the data view window. There's also some weird color bug that happened for the first time. It doesn't happen on every single transform, but it happens often enough that I tend to make one move, click ok, then transform again and make one move. This way I can always undo and not lose all of my progress.

https://github.com/user-attachments/assets/52a7329d-82e4-4ea7-adb8-78b57a3a6b08

JustTryingToGetSomeWorkDone, 2024-11-25:
Yep, it saved the damage, but took a couple of tries. The file size almost doubled but it's the same parts.
[recreateFreeCADbugSavedDamage_011425.zip](https://github.com/user-attachments/files/17897021/recreateFreeCADbugSavedDamage_011425.zip)

tarman3, 2024-11-25:
Can not reproduce.

<details>
  <summary>FreeCAD 1.0.0.39109 (Git), Arch Linux (KDE/plasma/wayland)</summary>

```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.31; sheetmetal 0.5.7; Silk 0.1.5; Curves 0.6.51
```
</details>

----------------------------------------------------------------------

<details>
  <summary>FreeCAD 1.0.0.39109 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.31; sheetmetal 0.5.8; Silk 0.1.5; Curves 0.6.51
```
</details>

----------------------------------------------------------------------

<details>
  <summary>FreeCAD 1.1.0dev.39240 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39240 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: f9ba94c70b139ece6ba5ffba47632d91b5c9c645
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.31; sheetmetal 0.5.7; Silk 0.1.5; Curves 0.6.51
```
</details>

JustTryingToGetSomeWorkDone, 2024-11-26:
Maybe someone running Windows 10? or at least 11?

PR #17564 (2025-01-13): Gui: New transform dialog
This refactors `ViewProviderDragger` and `TaskCSysDragger` to be more modern and to support selecting Transform Origin. It adds ability to snap transform origin, enter exact coordinates, snap one object to another

> [!CAUTION] 
> **Breaking Changes** 
>
> This removes support for `UseCenterballDragger` from Link view provider. this was obscure feature that was in core basically only because Assembly 3 used it. Keeping it would drastically complicate code and provide yet another place in code where something behaves differently.
>
> As far as I understand, it can be r [... cut]
closed 2025-01-13 by chennes