FreeCAD/FreeCAD #18026: Lattice2 ShapeString for Arraying error

marctremblay337, 2024-11-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Running the Python command 'Lattice2_ShapeString' failed

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Installed mods: 
  * A2plus 0.4.68
  * AirPlaneDesign 0.4.1
  * Assembly4 0.50.15
  * CurvedShapes 1.0.13
  * Curves 0.6.51
  * ExplodedAssembly
  * fasteners 0.5.31
  * fcgear 1.0.0
  * lattice2 1.0.0
  * LCInterlocking
  * Manipulator 1.5.7
  * Plot 2024.10.23
  * Pyramids-and-Polyhedrons
  * Render 2024.11.20
  * sheetmetal 0.5.6
  * woodworking 0.21.2.33771
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-21:
Please open an issue in the Lattice2 repository

closed 2024-11-21 by maxwxyz

luzpaz, 2024-11-22:
https://github.com/DeepSOIC/Lattice2