FreeCAD/FreeCAD #18027: v1.0 - Windows - Unnormalized Transient Dir

howetuft, 2024-11-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Starting with 1.0, transient directory returned by `App.ActiveDocument.TransientDir` on Windows platform mixes slashes and backslashes.
Typically:
`C:/Users/XXXXX/AppData/Local/cache\\FreeCAD\\Cache\\FreeCAD_Doc_244a255f-71e2-4596-8f17-6e46cb4c29f5_da39a3-452036`

This makes the path unusable for some external applications, for instance Pov-Ray. This affects Render Workbench.

Solution: the result of `App.ActiveDocument.TransientDir` should be normalized by FreeCAD, for instance by applying `os.path.normpath`.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US) [ OS: French/France (fr_FR) ]
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods:
  * A2plus 0.4.60m
  * Assembly3 0.12.0
  * Assembly4 0.50.12
  * fasteners 0.5.20
  * Movie 2024.2.15
  * Render 2024.9.25
  * Silk 0.1.2
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-21:
@wwmayer FYI

Roy-043, 2024-11-21:
Related:
https://github.com/FreeCAD/FreeCAD/issues/11970#issuecomment-2080108749

luzpaz, 2025-11-15:
@maxwxyz this seems like it's important. Maybe even worth backporting ?

maxwxyz, 2025-11-16:
is this still relevant?

FEA-eng, 2025-11-17:
> is this still relevant?

See my comment [here](https://github.com/FreeCAD/FreeCAD/issues/18027#issuecomment-3531527650)

luzpaz, 2026-10-08:
@FEA-eng can you retest to see if this is still relevant ?

FEA-eng, 2026-10-09:
I get this:

> 'C:\\Users\\username\\AppData\\Local\\Temp\\\\FreeCAD\\v27-1\\Cache\\FreeCAD_Doc_bb87df43-3b80-4644-b0e9-9737cc8c4686_da39a3_6682987655237592'

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261006 (Git shallow)
Build date: 2026/10/06 21:00:40
Build type: Release
Branch: (HEAD detached at 3424c4106)
Hash: 3424c410627263394f701aeec5b61d7bab7bb3a1
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 617.14/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```

maxwxyz, 2026-11-07:
Part of this is fixed in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/f74785e385862d165a5e846ae719c7718af4df96

The standard user paths use native separators before subdirectories are added. It addresses mixed slashes, but does not remove repeated separators mentioned later in the discussion.