FreeCAD/FreeCAD #18029: FreeCAD - CAM Face operation problem

Pendelf, 2024-11-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Operation Face doesn't work at all

https://github.com/user-attachments/assets/ef45ed37-90ac-422b-a398-0eee3a149609

[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.51
  * fasteners 0.5.31
  * PieMenu 1.9.5
  * sheetmetal 0.5.6
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dresco, 2024-12-14:
This might be a duplicate of #17748
@Pendelf - does it work if you change the Boundary Shape to FaceRegion instead of Boundbox?

Pendelf, 2024-12-14:
@dresco 
Yeah, then it works

[image]

Pendelf, 2025-09-01:
@maxwxyz 

In the latest version and in version 1.0.2, it works fine, so we can probably close it.

https://github.com/user-attachments/assets/d7322bc0-87a9-487a-9a3e-dd2104c914a2