FreeCAD/FreeCAD #18030: DesignSPHysics not compatible

robot21-5, 2024-11-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The DesignSPHysics can be installed but cannot be lauched as a compatibility arise: "This FreeCAD version is not compatible. Please update FreeCAD to version 018 or higher."

### Full version info

```shell
This FreeCAD version is not compatible. Please update FreeCAD to version 018 or higher.
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-21:
Please open an issue in the respective repository of the 3rd party addon.

closed 2024-11-21 by maxwxyz

luzpaz, 2024-11-21:
See:
* https://github.com/DualSPHysics/DesignSPHysics/issues/218