FreeCAD/FreeCAD #18032: TechDraw autofill not respecting locale for template date autofill

mpfusion, 2024-11-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

According to https://github.com/FreeCAD/FreeCAD/pull/13005 the autofilled date in the templates should respect the user's locale:

> The date should correspond to the current time zone (not a UTC date) and will be formatted by currently selected locale in the short format. 

However, that's not what happens, at least not in all cases.

To reproduce
===========
- set LC_TIME to ISO 8601:
  `LC_TIME=en_DK.UTF-8`
- confirm date is set to ISO 8601:
 ```
$ date
2024-11-21T09:52:44 CET
```
- open FreeCAD, create a simple part and open TD workbench
- click “Insert Page using Template”
- select “A4_LandscapeTD.svg”

→ date is formatted: 21/11/2014

[image]

Expected behaviour
================
Since the locale LC_TIME is set to `en_DK` (ISO 8601) I expect the date in TD to be formatted in ISO 8601 format (2024-11-21).

Full locale info:

```
$ locale
LANG=en_IE.UTF-8
LANGUAGE=
LC_CTYPE=en_IE.UTF-8
LC_NUMERIC="en_IE.UTF-8"
LC_TIME=en_DK.UTF-8
LC_COLLATE="en_IE.UTF-8"
LC_MONETARY="en_IE.UTF-8"
LC_MESSAGES="en_IE.UTF-8"
LC_PAPER="en_IE.UTF-8"
LC_NAME="en_IE.UTF-8"
LC_ADDRESS="en_IE.UTF-8"
LC_TELEPHONE="en_IE.UTF-8"
LC_MEASUREMENT="en_IE.UTF-8"
LC_IDENTIFICATION="en_IE.UTF-8"
LC_ALL=
```

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/lightdm-xsession/xcb)
Architecture: x86_64
Version: 1.1.0dev.39241 (Git)
Build type: Release
Branch: main
Hash: cdb92768fe4f33d900664415c613f9eced5b8927
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/Ireland (en_IE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.29
  * kicadStepUpMod 11.3.9
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-21:
@WandererFan can confirm, but only with the autofill command.
Workaround: right click on the field and choose autofill:
[image]

mpfusion, 2024-11-21:
I'm not sure what exactly you mean by autofill “command”. Autofill happens automatically when you insert a new page. No need to execute a dedicated “command”.

Notwithstanding, the workaround doesn't work for me. The “Change Editable Field” opens with a *left* click, rather than a *right* click on my system. When I click “Autofill” the date format is still wrong:
[image]

WandererFan, 2024-11-21:
forum discussion: https://forum.freecad.org/viewtopic.php?t=92238

WandererFan, 2024-11-23:
AFAICT, the only way to ensure that Qt formats the date as ISO8601 is to not use locale based date formatting, but to force the use of format "Qt::ISODate" and trim off the time part. 

This will require an "enforce ISO date format" preference that will override the locale format.  If you want ISO date format, tick the box, otherwise you get whatever Qt and the OS decide.

There are some hits on google for problems getting the right date format in Qt, but no solutions other than brute force as described above.

Qt's default locale is not the OS locale(!), but en_US (with some mods?).  It also appears that locale results can change based on OS or even compiler.

If somebody has a better solution I'm hapy to listen.

Some investigation results: 
If I force the locale to en_DK.UTF-8,  Qt tells me this:
11:30:20  DT::getAutofillValue - OS date formats for en_DK.UTF-8 - short: dd/MM/yyyy  long: dddd, d MMMM yyyy

If I force to da_DK, I get:
11:41:53  DT::getAutofillValue - OS date formats for da_DK - short: dd.MM.yyyy  long: dddd 'den' d. MMMM yyyy

Neither of which is ISO8601, and for both, the template shows the date formatted as in the short format.

In my case, with number format = Operating System, I get ccyy-mm-dd, because, surprisingly:
12:04:14  DT::getAutofillValue - OS date formats for en_US - short: yyyy-MM-dd  long: dddd, MMMM d, yyyy 

with number format = selected language, I still get ccyy-mm-dd, but:
12:08:13  DT::getAutofillValue - OS date formats for en_US - short: M/d/yy  long: dddd, MMMM d, yyyy

with C/POSIX, I get dd MMM, ccyyy, which isn't quite the short or long format.
12:11:48  DT::getAutofillValue - OS date formats for en_US - short: d MMM yyyy  long: dddd, d MMMM yyyy

mpfusion, 2024-11-23:
> Qt's default locale is not the OS locale(!), but en_US (with some mods?).

That sounds like a bug (in Qt) to me. What use is the OS locale if Qt programs can't use it?

> It also appears that locale results can change based on OS or even compiler.

This sounds even more like a bug to me. I cannot imagine that that behaviour is intentional.

> If somebody has a better solution I'm happy to listen.

Just a thought: Would it be possible to not use Qt to determine the locale? I mean `setlocale` provides the (correct) locale string without much effort:

```
#include <locale.h>
#include <stdio.h>
#include <stdlib.h>

int main(int argc, char *argv[])
{
   // complete locale
  // → LC_CTYPE=en_IE.UTF-8;LC_NUMERIC=en_IE.UTF-8;LC_TIME=en_DK.UTF-8;…
   printf("%s\n", setlocale(LC_ALL, ""));

   // only date formatting
  // → en_DK.UTF-8
   printf("%s\n", setlocale(LC_TIME, ""));

   exit(EXIT_SUCCESS);
}
```

closed 2024-12-13 by chennes

mpfusion, 2024-12-14:
Is this bug really fixed? It still doesn't work for me. Date is still showing as DD/MM/YYYY although my locale states YYYY-MM-DD:
[image]
````
$ date
2024-12-14T12:02:30 CET
````
Can someone please confirm if it's actually working?
````
OS: Debian GNU/Linux trixie/sid (X-Cinnamon/lightdm-xsession/xcb)
Architecture: x86_64
Version: 1.1.0dev.39646 (Git)
Build type: Release
Branch: main
Hash: b25b83c1a58bf0ae95cbd502d0ccae27e8127cad
Python 3.12.7, Qt 5.15.15, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * kicadStepUpMod 11.4.4
````
[datetest.tar.gz](https://github.com/user-attachments/files/18135539/datetest.tar.gz)