FreeCAD/FreeCAD #18035: Draft_Dimension add 'Annotation Style' option to the Task panel

semhustej, 2024-11-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When user is using Annotation Styles for managing dimension style, he/she firs has to create the dimension and than assign the Annotation style to the dimension in the properties of the dimension. This has several disadvantages:

- user has to repeatedly set the correct annotation style
- when the dimension is first created, the user has no overview how the dimension looks like in the correct style

**Proposed solution**

- It would be handy if 'Annotation style' with a drop down of options would be available in the Draft_Dimension task panel.

[image]

- The setting should be preserved after the Draft_Annotation command has been invoked again
- In the new file (or on FreeCAD startup), the dropdown option should default to 'None' or 'Default' value, which is how Draft_Diemnsion is currently working

It would be nice, if the same was applied to all tools, which create objects with 'Annotation Style' parameter:

- Draft_Dimension
- Draft_Text
- Draft_Label

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US) [ OS: English/United States (en_US) ]
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

Draft

### Anything else?

**Alternative solution**
Create a new option to set 'Active Annotation Style' in the Draft_AnnotationStyleEditor

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-11-23:
@Roy-043 thoughts ?

Roy-043, 2024-12-04:
IMO this is not a good idea. It would be better to have an active annotation style instead.

yorikvanhavre, 2024-12-05:
It is true that it would go against all the way Draft works... Then there will be no reason why that does not apply to other Draft types, and then the whole Draft current style system would become obsolete.

Maybe what we would need instead is focusing on how quick and easy one can 1) select an active style and 2) change the style of an object

semhustej, 2024-12-05:
> IMO this is not a good idea. It would be better to have an active annotation style instead.

I thought the solution I proposed would be easier to implement. But if setting active style is possible to add to Annotation styles editor, that is even better solution than the original proposition.

semhustej, 2024-12-05:
> 
>  2) change the style of an object

I am not sure what you have in mind, but I think current way of selecting object/objects and changing the annotation style in properties from the drop down list is good enough.

yorikvanhavre, 2024-12-06:
> I am not sure what you have in mind, but I think current way of selecting object/objects and changing the annotation style in properties from the drop down list is good enough.

I agree, hence the need for something better. Not sure what form it should take, though