FreeCAD/FreeCAD #18036: Freecad on Garuda/ Arch work area blank.

prevail90, 2024-11-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I open a project either new or a work in progress one, as you see below the area is blank. I have tried it in both normal mode and safe mode in both dark and light mode. I have tried both the appimage which doesnt even open and using my system's app installer which atleast freecad opens.

[image]

### Full version info

```shell
FREECAD 
OS: Garuda Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * 3D_Printing_Tools
  * Assembly4 0.50.15
  * Gridfinity 0.6.0
  * Lithophane
  * sheetmetal 0.5.6
  * InventorLoader 1.4.0
  * offline-documentation 1.0.0-alpha
```

### Subproject(s) affected?

None

### Anything else?

```text
System:
  Kernel: 6.12.0-arch1-1-znver4 arch: x86_64 bits: 64 compiler: gcc v: 14.2.1 clocksource: tsc
    avail: hpet,acpi_pm parameters: BOOT_IMAGE=/@/boot/vmlinuz-linux-znver4
    root=UUID=aec83d4b-e37e-4f72-b6ea-5c4d04c66879 rw rootflags=subvol=@ quiet
    resume=UUID=68203e8d-b3d1-404d-b55c-2c324445eb76 loglevel=3 ibt=off
  Desktop: KDE Plasma v: 6.2.3 tk: Qt v: N/A info: frameworks v: 6.8.0 wm: kwin_wayland vt: 1
    dm: SDDM Distro: Garuda base: Arch Linux
Machine:
  Type: Desktop Mobo: ASUSTeK model: G10DK v: 1.0 serial: <superuser required>
    uuid: <superuser required> UEFI: ASUSTek v: 310 date: 09/29/2022
CPU:
  Info: model: AMD Ryzen 7 5700G with Radeon Graphics bits: 64 type: MT MCP arch: Zen 3 gen: 3
    level: v3 note: check built: 2021-22 process: TSMC n7 (7nm) family: 0x19 (25) model-id: 0x50 (80)
    stepping: 0 microcode: 0xA50000C
  Topology: cpus: 1x dies: 1 clusters: 1 cores: 8 threads: 16 tpc: 2 smt: enabled cache:
    L1: 512 KiB desc: d-8x32 KiB; i-8x32 KiB L2: 4 MiB desc: 8x512 KiB L3: 16 MiB desc: 1x16 MiB
  Speed (MHz): avg: 400 min/max: 400/5288 boost: enabled scaling: driver: amd-pstate-epp
    governor: performance cores: 1: 400 2: 400 3: 400 4: 400 5: 400 6: 400 7: 400 8: 400 9: 400
    10: 400 11: 400 12: 400 13: 400 14: 400 15: 400 16: 400 bogomips: 137656
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm
  Vulnerabilities: <filter>
Graphics:
  Device-1: NVIDIA GA106 [GeForce RTX 3060 Lite Hash Rate] vendor: ASUSTeK driver: nvidia
    v: 565.57.01 alternate: nouveau,nvidia_drm non-free: 550.xx+ status: current (as of 2024-09;
    EOL~2026-12-xx) arch: Ampere code: GAxxx process: TSMC n7 (7nm) built: 2020-2023 pcie: gen: 3
    speed: 8 GT/s lanes: 16 link-max: gen: 4 speed: 16 GT/s ports: active: none
    off: DP-2,DP-3,HDMI-A-1 empty: DP-1 bus-ID: 01:00.0 chip-ID: 10de:2504 class-ID: 0300
  Device-2: Advanced Micro Devices [AMD/ATI] Cezanne [Radeon Vega Series / Radeon Mobile Series]
    vendor: ASUSTeK driver: amdgpu v: kernel arch: GCN-5 code: Vega process: GF 14nm built: 2017-20
    pcie: gen: 3 speed: 8 GT/s lanes: 16 link-max: gen: 4 speed: 16 GT/s ports: active: none
[... 81 lines cut ...]
  ID-2: swap-2 type: zram size: 46.35 GiB used: 5.7 MiB (0.0%) priority: 100 comp: zstd
    avail: lzo-rle,lzo,lz4,lz4hc,deflate,842 max-streams: 16 dev: /dev/zram0
Sensors:
  System Temperatures: cpu: 48.6 C mobo: N/A gpu: amdgpu temp: 39.0 C
  Fan Speeds (rpm): cpu: 900
Info:
  Memory: total: 48 GiB note: est. available: 46.35 GiB used: 4.54 GiB (9.8%)
  Processes: 426 Power: uptime: 16h 30m states: free
ze,mem,disk suspend: deep avail: s2idle
    wakeups: 0 hibernate: platform avail: shutdown, reboot, suspend, test_resume image: 18.49 GiB
    services: org_kde_powerdevil, power-profiles-daemon, upowerd Init: systemd v: 256
    default: graphical tool: systemctl
  Packages: pm: pacman pkgs: 2177 libs: 548 tools: octopi,paru Compilers: clang: 18.1.8
    gcc: 14.2.1 Client: shell wrapper v: 5.2.37-release inxi: 3.3.36
Garuda (2.6.26-1):
  System install date:     2024-03-17
  Last full system update: 2024-11-21 ↻
  Is partially upgraded:   No
  Relevant software:       snapper NetworkManager dracut nvidia-dkms
  Windows dual boot:       No/Undetected
  Failed units:            

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-21:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/6177

prevail90, 2024-11-26:
this is not a duplicate of #6117 as I cannot see anything on the workbench. no matter what workbench i am in. I have also tried multiple different kernel versions with no avail.

prevail90, 2024-11-26:
The issue is wayland and it not talking to the depricated x11 as easy.

darrellenns, 2025-02-17:
Same issue here on Arch Linux with nvidia and gnome wayland. Several others report the same issue in this forum thread: https://forum.freecad.org/viewtopic.php?style=3&t=89408

Setting QT_QPA_PLATFORM=xcb does work around the issue.