FreeCAD/FreeCAD #18037: 1.0.0 on Fedora 41 fails at launch.

hickscorp, 2024-11-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The 1.0.0 release doesn't seem to work with Fedora 41.

### Full version info

```shell
Upon launching, it asks whether I would like to "Perform system integration" (Which fails with `Failed to register AppImage in system via libappimage`) or "Run Once". Regardless, it crashes right after that with:

./FreeCAD_1.0.0-conda-Linux-aarch64-py311_ea574ddcfb32b30db251188b4ab190bf.AppImage
/usr/bin/AppImageLauncher: /lib64/libcurl.so.4: no version information available (required by /usr/bin/../lib/x86_64-linux-gnu/appimagelauncher/libappimageupdate.so)
Squashfs image uses (null) compression, this version supports only xz, zlib.
ERROR: appimage_shall_not_be_integrated : sqfs_open_image error: /home/doodloo/.local/Applications/FreeCAD_1.0.0-conda-Linux-aarch64-py311_ea574ddcfb32b30db251188b4ab190bf.AppImage
AppImageLauncher error: appimage_shall_not_be_integrated() failed (returned -1)
Squashfs image uses (null) compression, this version supports only xz, zlib.
ERROR: appimage_is_terminal_app : sqfs_open_image error: /home/doodloo/.local/Applications/FreeCAD_1.0.0-conda-Linux-aarch64-py311_ea574ddcfb32b30db251188b4ab190bf.AppImage
AppImageLauncher error: appimage_is_terminal_app() failed (returned -1)
[appimage-binfmt-bypass/bin] ERROR: failed to execute patched runtime: No such file or directory
```

Note that the RC2 AppImage used to work fine.
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hickscorp, 2024-11-21:
Erroreous report. Sorry!

closed 2024-11-21 by hickscorp

davekch, 2025-02-09:
hi @hickscorp , I'm hitting the same issue and it seems like you were able to easily resolve this yourself. do you mind sharing what's the trick here? would appreciate it, thanks!