FreeCAD/FreeCAD #18038: 1.0.0 AppImage Fails to launch on Fedora 41.

hickscorp, 2024-11-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

./FreeCAD_1.0.0-conda-Linux-x86_64-py311_762d03e714090fee85c2cd69af722050.AppImage
/usr/bin/AppImageLauncher: /lib64/libcurl.so.4: no version information available (required by /usr/bin/../lib/x86_64-linux-gnu/appimagelauncher/libappimageupdate.so)
Squashfs image uses (null) compression, this version supports only xz, zlib.
ERROR: appimage_shall_not_be_integrated : sqfs_open_image error: /home/doodloo/Downloads/FreeCAD_1.0.0-conda-Linux-x86_64-py311_762d03e714090fee85c2cd69af722050.AppImage
AppImageLauncher error: appimage_shall_not_be_integrated() failed (returned -1)
Squashfs image uses (null) compression, this version supports only xz, zlib.
ERROR: appimage_is_terminal_app : sqfs_open_image error: /home/doodloo/Downloads/FreeCAD_1.0.0-conda-Linux-x86_64-py311_762d03e714090fee85c2cd69af722050.AppImage
AppImageLauncher error: appimage_is_terminal_app() failed (returned -1)
execv error: No such file or directory

### Full version info

```shell
1.0.0 AppImage - but can't open "About" dialog as it won't launch at all.
```

### Subproject(s) affected?

None

### Anything else?

RC2 AppImage works fine.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-11-21:
Don not use appimagelauncher.
Try with command in terminal: `./ FreeCAD_1.0.0-conda-Linux-aarch64-py311.AppImage `

maxwxyz, 2024-11-21:
see above

closed 2024-11-21 by maxwxyz