FreeCAD/FreeCAD #18045: Cannot set link transparency when selecting it from the Objects Tree

leoheck, 2024-11-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Cannot set the transparency of a Link when selecting it from the Objects Tree (using `V,T` shortcut and also in the context menu). However, it works when I click the Object itself.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.39240 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: f9ba94c70b139ece6ba5ffba47632d91b5c9c645
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.4.26
  * InventorLoader 1.4.0
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.0.24
  * lattice2 1.0.0
  * kicadStepUpMod 11.3.9
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2026-10-17:
The original issue seems solved, but @leoheck mentioned further issues, so we may wait for their confirmation.

leoheck, 2026-10-17:
@FEA-eng, thanks I can open a new issue with extra info if I find something related in the future.

closed 2026-10-17 by leoheck