FreeCAD/FreeCAD #18046: Draft Scale and Snap widgets are not named in toolbar lists

semhustej, 2024-11-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

- open new file with Draft WB selected
- right click on one of the toolbars
- Scale Widget and Snap Widget are possible to turn off, but have no names. See screenshot:

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-11-21:
Related/duplicate: #16832.

semhustej, 2024-11-21:
Closing this issue as a duplicate.

closed 2024-11-21 by semhustej