FreeCAD/FreeCAD #18047: Part: boolean operation fails with wire-offset-extruded solid

steel-ne, 2024-11-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:
- Part: construct solid with a hole (sketch+extrude)
- Part: select face with hole, click "Create shape element copy"
- Draft: Downgrade copied face to wires
- Part: Select wire of hole, apply 2D offset -2mm (inside a hole), select "Fill offset"
- Part: Extrude offset to 10 mm
- Part: Create simple shape cube, allocate it to intersect with the extrude
- Try any boolean (cut, union, etc), found error in a console: "Cut: A fatal error occurred when running boolean operation"
[image]
[image]
[image]

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * Assembly4 0.50.13
  * Curves 0.6.39
```

### Subproject(s) affected?

None

### Anything else?

File to demonstrate a problem. Open and try to cut Box and Extrude001
[boolean-problem.zip](https://github.com/user-attachments/files/17850134/boolean-problem.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-11-21:
Offset2D with Fill=true is broken

steel-ne, 2024-11-21:
Definitely. If I workaround it with "Offset2D" -> "Part_MakeFace" problem is gone.

steel-ne, 2024-11-21:
I can not find an issue that Offset2D with Fill is broken, leave it open

tarman3, 2024-11-21:
I think the problem is a combination of conditions:
- Create shape element copy
- Offset2D with Fill

CoreyBurger, 2025-09-03:
I can still replicate this with FreeCAD 1.0.2 on Windows (I was the original filer of #19054)