FreeCAD/FreeCAD #18049: Changing Freecad layout with F3 and F4 is annoying.

leoheck, 2024-11-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Please, remove the F3, F4 shortcuts, they make the whole use of Freecad impossible.
The standard shortcut to rename things is F2 (on most of OSs) then we also have the special function keys on those keys. 

Changing the layout of Freecad should be done once and not frequently. But renaming objects happens all the time.
Right now, it is impossible to work with Freecad without having the layout messed up constantly. 

Please, fix this annoying issue!

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.39240 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: f9ba94c70b139ece6ba5ffba47632d91b5c9c645
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.5.6
  * InventorLoader 1.4.0
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.31
  * osh-autodoc-workbench 0.2.3
  * OpenTheme.backup1732209044.7838953 2024.9.1 (Disabled)
  * OpticsWorkbench 1.0.26
  * lattice2 1.0.0
  * kicadStepUpMod 11.4.1
```

### Subproject(s) affected?

Core

### Anything else?

This should be fixed on 1.1 and also backported to 1.0. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

leoheck, 2024-11-21:
Also, shortcuts to change the view, 1, 2, 3, are right under the F2, F3, F4 on most of the keyboards.

alexneufeld, 2024-11-21:
agreed. note also that F5 is recompute, so it's easy to accidentally press F4 while trying to refresh the document. Just removing the F3/F4 default keybindings seems reasonable to me

leoheck, 2024-11-21:
I think that it is also possible to harden the shortcuts with Ctrl or Shit or something else that is not immediately available to be accessed by mistake since the operation on F3 and F4 takes time to be executed and also takes time to be reset since the operation on F3 and F4 is a list of different setups.

obelisk79, 2024-11-22:
I think either adding a modifier CTRL-F3/F4 is acceptable.

semhustej, 2024-11-22:
> I think either adding a modifier CTRL-F3/F4 is acceptable.

Ctrl+F4 closes the active tab. Additionally this is standard key binding across many software solutions.

obelisk79, 2024-11-22:
I'm not even sure a change is really necessary here. I use F2 for renaming things very frequently and have not once run into this *issue*.

leoheck, 2024-11-22:
> I'm not even sure a change is really necessary here. I use F2 for renaming things very frequently and have not once run into this _issue_.

This means you don't use computers/keyboards as others use. There are some people that sits in front of the screen staring at it and barely touch the keyboard.

obelisk79, 2024-11-22:
> This means you don't use computers/keyboards as others use. There are some people that sits in front of the screen staring at it and barely touch the keyboard.

As someone who relies heavily on keyboard shortcuts in most software that I use, I know the difference. Not sure what you're trying to say here.

evan-king, 2025-01-19:
I hesitate to put words in @leoheck's mouth, but suspect the underlying point driving their response is how unhelpful "git gud at keyboard" is, compared to reasonable and very easy changes that can limit the destructiveness of accidental inputs.

I generally favor keyboard-first behavior, but it feels like ofttimes FreeCAD actively discourages me from investing in that proficiency.  (I'm looking at you: function edit modals behind tiny icon with no hotkey; variable autocomplete dropdowns with no tab navigation making me jump between home row and arrow keys; and non-standard edge/vertex selection behavior favoring mouse-only input at the expense of error-prone selection clearing).

I post this comment moments after failing to find a way to disable these hotkeys, because I'm tired of hosing my UI layout while fat-fingering scene updates on a keyboard whose function keys are obscured by desk overhang.  The hotkeys would be fine, if these "toggles" actually only did what they claim instead of also resetting layout.  It took me a bit to discover how I was even doing it, because it always happened while I was on a roll and watching only the screen.

**Workaround**:

Force-quitting the app prevents the layout changes from being saved by a clean shutdown (on Windows at least) for which I am immensely thankful.

loughkb, 2025-02-08:
I would also like to vote for f3 and F4 to be changed with a modifier or disabled. Several times I've accidentally hit f3 when trying to rename something and then spent almost minutes trying to get my user interface fixed again. Very annoying. And I just saw a post over on the freecad Facebook page from a guy who accidentally hit F4 and is struggling to get his interface fixed. Text colors changed in the tree and went invisible. I can't even find what F4 is supposed to do. But it certainly destroyed his interface.

JohnOCFII, 2025-02-08:
Related -- a solid Undo/Redo to help undo accidental UI changes would be appreciated.

tarman3, 2025-02-08:
[image]

loughkb, 2025-02-08:
Thanks tarman,  I also found that this morning... I have cleared them both.

Interesting though that they are labeled as "toggle" overlay when they don't .  When I accidentally hit F3 once, I immediately hit it again, but it didn't change back.  The fellow that posted to the facebook page about hitting F4 also tried hitting it again, but it didn't change back.

obelisk79, 2025-02-08:
These 'functions' probably shouldn't be assigned a default hot key at all considering their behavior is so illogical. That is probably the most "low hanging fruit" solution to this.

leoheck, 2025-02-19:
Now this `F1` is annoying my soul out. 

It is horrible to mix functions that take time to load (e.g., the help menu that opens the browser and freezes Freecad while it is opening) with tasks that should take action instantly, such as renaming objects.

- F1 (UI freezes while launching the browser and changes the window focus once it is done)
- F2 (quick action to rename)
- F3 (change the whole UI layout which takes time)
- F4 (change the whole UI layout which takes time)

`F2` is hijacked in between those horrible to-use-and-slow to-perform functionalities.

This is how the keyboard is laid out if someone does not know yet.
``` ESC | F1 | F2 | F3 | F4 ```

And it varies from computer to computer but F-Keys are small and mostly squished in a tight space.

b3nsn0w, 2025-06-11:
not sure how this issue going but i'd just like to bump it and voice my support. having a quick action and one that completely messes up the ui with no quick way to restore it literally one hotkey away is _horrible_ ux. i can't put into words how much frustration and broken flow this has caused for me, and i'm sorry but "just don't mess up, see i didn't mess up" is not an acceptable answer. it should be freecad's responsibility to make it as difficult to mess up as possible, especially on something as basic as the ui.

refinements like this would make the difference between a niche tool by programmers for programmers only vs a respectable software that you can recommend to non-technical users. by the sounds of it this should be a quick and easy change, so why not implement it?

i assume this would be two lines in a descriptor file. the only reason i don't wanna open a pull request is i'm concerned it would devolve into bikeshedding, but please do something about it.

for a specific option, i agree with @tarman3's and @obelisk79's solution, just unbind the shortcuts. the approximately 3 users who use F3/F4 on a daily basis could always just rebind it. for everyone else, you'd gain two valuable easy-to-press hotkeys for actually useful functions that could further improve freecad's ux, while also eliminating a source of frustration.

but that's just one way to fix it. honestly, i don't care which path you take, just _please_ take one. it should not be the user's responsibility to fix the software.

leoheck, 2025-11-13:
No,  it won't. You are trying to solve a different problem. And still, your suggestion, is good for that different problem.

But the issue here is "the waist of time" (times 2) we experience when we hit this key by accident. The action takes 1 or 2s and it changes the whole screen. And this happens very often since the key next to it is a very important hotkey. It is the standard hotkey for renaming things.

furgo16, 2025-11-19:
I just stumbled upon this issue. I don't have a stron opinion on this, I'll just add some data points. These are the commands these shortcuts trigger, since as far as I can gather, they've not been explicitly mentioned:

- <kbd>F3</kbd>: https://wiki.freecad.org/Std_DockOverlayToggle
- <kbd>F4</kbd>: https://wiki.freecad.org/Std_DockOverlayAll

leoheck, 2025-11-20:
Adding another hotkey to those keys will significantly improve this behavior or fix this issue, as those functions won't be triggered by accident.

leoheck, 2026-02-24:
Thanks @semhustej. 
I was traveling, but I’ve now also confirmed that the annoying behavior of the F3/F4 keys is gone.

```
OS: Ubuntu 25.10 (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.2.0dev.20260217 (Git shallow)
Build date: 2026/02/17 21:26:14
Build type: Release
Branch: (HEAD detached at 3bad5dabd)
Hash: 3bad5dabde38f1a901fa1475e30f5caee7f9846a
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/144.289/1
```