FreeCAD/FreeCAD #18050: Hidden line colour in techdraw always black even when changed in preferences.

Suthy1, 2024-11-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The hidden line colour option in techdraw does not work on hidden lines in techdraw drawings in FC 1.0. This previously worked fine in FC 0.21.2.

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/New Zealand (en_NZ)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * ArchTextures
  * BillOfMaterials 0.1.4.2
  * CADExchanger
  * CurvedShapes 1.0.9
  * Curves 0.6.46
  * dxf-library
  * fasteners 0.5.28
  * parts_library
  * QuickMeasure 2022.10.28
  * sheetmetal 0.4.24
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-11-21:
Can you [restart in Safe Mode](https://wiki.freecad.org/Std_RestartInSafeMode) and see if you can reproduce ?

Suthy1, 2024-11-21:
Yes, the issue is still present.
[image]
[image]

Suthy1, 2024-11-22:
Looking into this further, the hidden line colour seems to be linked with the "Normal" line colour.
[image]
[image]

commit 0f6809f3 (2024-12-09): [TD]fix #18050 hidden line color
closed 2024-12-09 by yorikvanhavre