FreeCAD/FreeCAD #18051: STL Export displaces bodies / does not respect body placement coordinates

PM490, 2024-11-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Creating two bodies, and positioning them using the body placement generates is not maintained when exported to STL.

This may be related to [this closed issue](https://github.com/FreeCAD/FreeCAD/issues/12278), but since it was closed last week, would like to report it, and is repeatable.

Steps to repeat.

Create a part, inside it create a body, inside it create a sketch and draw a circle, then extrude into a cilinder of 10 mm (or whatever you wish).
Copy the body and move it into the same part.
Use the placement  z property to displace the copied cilinder in the z axis by 10mm (or whatever you wish).
Highlight the part and export an STL.

When the STL is imported into -for example- Prusa Slicer, rather than the expected cylinder, there will be two with an airgap between them.

Thank you for looking into this issue.

[image]

### Full version info

```shell
[code]
OS: KDE Flatpak runtime (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at b9bfa5c)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.7, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

Exporting STLs

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-22:
Please test on latest 1.0 version

tarman3, 2024-11-22:
Can not confirm with FreeCAD 1.0
Check with PrusaSlicer 2.8.1 and Blender 4.3.0

tarman3, 2024-11-22:
PrusaSlicer can import STEP files.
So if for some reason you still want to use old version of FreeCAD, you can try export models to STEP.