FreeCAD/FreeCAD #18052: Refine in part workbench not behaving/working as it previously would.

Suthy1, 2024-11-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Refining an object after a boolean operation in the part workbench is not working like it did in FC 0.21.2.

I've attached the same model in both FC 0.21.2 & 1.0 which have a seem down the centre of the bottom of the model.

It refines fine in 0.21 but refuses to in 1.0. This is important so these parts can be flattened in the sheet metal workbench.

Is this a regression or just poor modelling?

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/New Zealand (en_NZ)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * ArchTextures
  * BillOfMaterials 0.1.4.2
  * CADExchanger
  * CurvedShapes 1.0.9
  * Curves 0.6.46
  * dxf-library
  * fasteners 0.5.28
  * parts_library
  * QuickMeasure 2022.10.28
  * sheetmetal 0.4.24
```

### Subproject(s) affected?

Part

### Anything else?

[GUTTER REV A.zip](https://github.com/user-attachments/files/17852164/GUTTER.REV.A.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Mauricio0264, 2024-11-21:
Hola puedo ayudar

tarman3, 2024-11-22:
There is a lot of stuff in your file. Could you remove the unnecessary stuff so that it becomes clear what to pay attention to?

Suthy1, 2024-11-22:
Apologies, I've made the files a lot more simple and attached them.

I've recreated the problem in 1.0 and it will not perform the last boolean cut operation without the model disappearing with an error.
[image]

I loaded the same model into 0.21.2 and the boolean cut worked and it refined with no issue.
[image]

[GUTTER REV A SIMPLE.zip](https://github.com/user-attachments/files/17865432/GUTTER.REV.A.SIMPLE.zip)

Unfortunately I cant reproduce the original seam that wouldn't refine and the cut wont work in 1.0 but this snip shows the seam in the centre at the bottom.
[image]

tarman3, 2024-11-22:
Something wrong with **GUTTER**
[image]

Suthy1, 2024-11-22:
Only in 1.0 - The same part has no errors in 0.21.2.
[image]
I think the issue comes from the sheetmetal aspect, not the boolean/refine so I'll close this thread.

closed 2024-11-22 by Suthy1