FreeCAD/FreeCAD #18053: FreeCAD 1.0 Assembly BOM generation failures

wooddragon48, 2024-11-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Create Bill of Materials tool for the Assembly workbench only creates an empty spreadsheet regardless of the parts in the assembly. After saving, closing, and reopening the file the spreadsheet has a single column labeled "Index". See attached images for the before and after saving view of the spreadsheet.

I have attached a exceedingly simple file that exhibits the problem.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * 3D_Printing_Tools
  * A2plus 0.4.60i
  * animation
  * ArchTextures
  * Assembly4 0.12.5
  * CurvedShapes 1.0.4
  * Curves 0.6.5
  * dxf_library
  * ExplodedAssembly
  * fasteners 0.4.54
  * fcgear
  * freecad.gears 1.0.0
  * Info
  * kicadStepUpMod 10.16.8
  * Manipulator 1.5.0
  * OpenTheme 2024.9.1
  * pcb
  * Render 2023.2.5
  * SelectorToolbar
  * sheetmetal 0.2.61
  * ShortCuts
```

### Subproject(s) affected?

Assembly

### Anything else?

[test file.zip](https://github.com/user-attachments/files/17853534/test.file.zip)
[image]
[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-22:
Can confirm:
[image]

@PaddleStroke @adrianinsaval @chennes this is the same error as the issue which was never found

EvansMike, 2024-12-08:
Same fault with Linux a using FreeCAD_1.0.0-conda-Linux-x86_64-py311.AppImage

module 'FreeCADGui' has no attribute 'doCommandEval'Running the Python command 'Assembly_CreateBom' failed:
Traceback (most recent call last):
  File "/tmp/.mount_FreeCABhmecL/usr/Mod/Assembly/CommandCreateBom.py", line 94, in Activated
    self.panel = TaskAssemblyCreateBom()
                 ^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCABhmecL/usr/Mod/Assembly/CommandCreateBom.py", line 146, in __init__
    self.createBomObject()
  File "/tmp/.mount_FreeCABhmecL/usr/Mod/Assembly/CommandCreateBom.py", line 337, in createBomObject
    self.bomObj = Gui.doCommandEval("bomObj")
                  ^^^^^^^^^^^^^^^^^

module 'FreeCADGui' has no attribute 'doCommandEval'

IgorokNerovnyi1980, 2024-12-23:
Now it s  reproduce  for Mac M1 2020. 

12:41:45  (qt.qpa.fonts) Populating font family aliases took 58 ms. Replace uses of missing font family "Courier" with one that exists to avoid this cost. 
12:41:46  Migrating Start Workbench to Start command... done.
12:43:10  Running the Python command 'Assembly_CreateBom' failed:
Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Assembly/CommandCreateBom.py", line 94, in Activated
    self.panel = TaskAssemblyCreateBom()
                 ^^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Assembly/CommandCreateBom.py", line 146, in __init__
    self.createBomObject()
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Assembly/CommandCreateBom.py", line 337, in createBomObject
    self.bomObj = Gui.doCommandEval("bomObj")
                  ^^^^^^^^^^^^^^^^^

module 'FreeCADGui' has no attribute 'doCommandEval'Running the Python command 'Assembly_CreateBom' failed:
Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Assembly/CommandCreateBom.py", line 94, in Activated
    self.panel = TaskAssemblyCreateBom()
                 ^^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Assembly/CommandCreateBom.py", line 146, in __init__
    self.createBomObject()
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Assembly/CommandCreateBom.py", line 337, in createBomObject
    self.bomObj = Gui.doCommandEval("bomObj")
                  ^^^^^^^^^^^^^^^^^

module 'FreeCADGui' has no attribute 'doCommandEval'

H-Tsubota, 2025-01-23:
I have same problem and do some research.

This issue is same as #18615 and the problem (i.e. lack of "doCommandEval") is fixed with commit "d1707fdf10e0468155bf57ace64e219e2ab17b7e" at 2024/12/20.