FreeCAD/FreeCAD #18054: Fillet and chamfer can't be done when touching another face.

rcfocus, 2024-11-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If I try to do a chamfer on the edge as shown in the figure, it can be done only when the chamfer size is less than the height from between the face and the top face. In this example, the height is 10mm. The chamfer can be done only if the size < 10mm, e.g. 5mm or 9.99mm. If size is equal to 10mm then it can not be done.

The issue also occurs when I try to do a chamfer on edge of the top face. It occurs when the chamfer "touches" the middle face. I also tested fillet, it has the same issue.

I think it is a bug of comparison, " <= " should be used instead of " < ", somewhere in the program.

[image]

### Full version info

```shell
OS: macOS 15.1.1
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Dark theme/Qt default
Installed mods: 
  * Behave-Dark-Colors 0.0.1
  * ProDarkThemePreferencePack 1.0.0
  * Help 1.0.0-alpha
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-11-22:
Fillet radius 10mm will create null-area face.

maxwxyz, 2024-11-22:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/5561

closed 2024-11-22 by maxwxyz