FreeCAD/FreeCAD #18055: Property View: Variable with value set to the Label of another object doesn't update properly

SpeeQz1, 2024-11-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This is a bit of a specific problem so I will explain it step by step.

1. Create a `Body`
2. Then a `Body001`
3. Rename `Body001` to `MyBody`
4. Add a custom variable to `Body` named `Test` in group `Data` of type `App::PropertyString`
5. Open the expression editor for `Test` and type `<<MyBody>>`
6. The value that would be displayed for `Test` is `MyBody`
7. Rename `MyBody` to `MyBody1`
8. The variable `Test` won't update its value and it will show `MyBody` until the expression is manually reset, restarting the program does not fix this issue.

[image]

### Full version info

```shell
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * fasteners 0.5.31 (Disabled)
  * lattice2 1.0.0
  * Manipulator 1.5.7
  * parts_library
  * Alternate_OpenSCAD 1.0.0
  * CurvedShapes 1.0.13
  * FreeCAD-Ribbon 1.4.1
  * SearchBar 1.0.2
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Expressions

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mnesarco, 2024-11-22:
`<<MyBody>>` is wrong, use `<<MyBody>>.Label`

SpeeQz1, 2024-11-22:
My problem is with the different behaviour between `<<MyBody>>` and `<<MyBody>>.Label` upon reload. The first time detection of the value is the same for both.
1.
* `Test`: `<<MyBody>>.Label` => `MyBody`
* Rename `MyBody` to `MyBody1`
* Recompute
* `Test`: `<<MyBody>>.Label` => __`MyBody1`__
2.
* `Test`: `<<MyBody>>` => `MyBody`
* Rename `MyBody` to `MyBody1`
* Recompute
* `Test`: `<<MyBody>>` => __`MyBody`__

mnesarco, 2024-11-22:
It is because first time <<MyBody>>.Name is equal to <<MyBody>>.Label but after you change Label they are not the same obviously.

When you use the expression <<MyBody>> as the value of a String property (you should not) it is returning the object not the Label.  You must use the explicit thing that you want to get from the object. either .Name or .Label

I understand that you expect that <<MyBody>> returns the same as <<MyBody>>.Label but it is not the case.

maxwxyz, 2024-11-22:
I'm closing this. If there is an actual bug, feel free to comment and reopen.

closed 2024-11-22 by maxwxyz