FreeCAD/FreeCAD #18072: Additive helix issue

Andrethetech, 2024-11-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Additive helix around a cylinder fails with "AdditiveHelix: Tool shape is not valid for boolean operation" unless I set the Cone angle to 0.01 instead of 0.00 and that is unexpected behavior. I increased the tolerance and it had no effect also turned off refine .

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

PartDesign

### Anything else?

[image]
[leadscrew.zip](https://github.com/user-attachments/files/17872063/leadscrew.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-11-22:
This file is trying to burn my laptop ))

maxwxyz, 2024-11-22:
@FlachyJoe @AIRCAP FYI

Andrethetech, 2024-11-22:
Reduced from 100mm to 50mm , takes about 10-15 sec to compute on my system.
[leadscrew (2).zip](https://github.com/user-attachments/files/17872997/leadscrew.2.zip)

tarman3, 2024-11-22:
My laptop not so powerful, only 8 gb of RAM and this is not enougth or this file.

Andrethetech, 2024-11-22:
Ok I will try to reduce it as as I can but since the pitch is 4mm it can't be 5mm , I tried smaller values and I get weird results.

tarman3, 2024-11-22:
You are wrong.
Length 5 mm. Pitch 10 mm.
[image]

Andrethetech, 2024-11-22:
[leadscrew (3).zip](https://github.com/user-attachments/files/17873262/leadscrew.3.zip)
My bad I was actually capable to redue it to 5mm ,see the attached file

tarman3, 2024-11-22:
Wow. It is strange that the geometry does not contain errors when constructing a spiral in this way.
Try changing your approach.

[image]

tarman3, 2024-11-22:
To get double thread use **PolarPattern**
[image]

Andrethetech, 2024-11-22:
I changed the sketch and now it works with the cone angle set to 0.00 and consumes less memory .

Andrethetech, 2024-11-22:
Yes by changing the sketch.
[image]

Andrethetech, 2024-11-22:
Yes weird issues with helix and I ran into a issue where I was hitting 100% memory usage and I have 48Gb of ram at some point generating a 500mm leadscrew model.

FlachyJoe, 2024-11-22:
You can increase Deviation and AngularDeflection view properties of the body in order to save time and resources.

maxwxyz, 2024-12-17:
Is this still an issue in the latest weekly?