FreeCAD/FreeCAD #18075: Toggling trasnparency resets colors..

leoheck, 2024-11-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The same issues with the colors... 

I am setting diffuse color.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.5.6
  * InventorLoader 1.4.0
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.31
  * osh-autodoc-workbench 0.2.3
  * OpenTheme.backup1732209044.7838953 2024.9.1 (Disabled)
  * OpticsWorkbench 1.0.26
  * lattice2 1.0.0
  * kicadStepUpMod 11.4.1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-11-22:
> The same issues with the colors...
I am setting diffuse color.

Please elaborate. Are you referencing another specific issue? This is too sparse.

tarman3, 2024-11-22:
Can not confirm. Write step to reproduce.

Looks like fine
https://github.com/user-attachments/assets/9214bd0c-6e50-4b77-9938-fa2d3135643d

leoheck, 2024-11-22:
1. Select an object (I am using part design body)
2. Ctrl+D, Set colors... I am using Diffuse and Blue.
3. Toggle transparency, it turns back gray.
4. Toggle transparency again, it is still gray... 

Note: I am using OpenTheme Dark ` 2024.9.1`

When I was writing this, the colors were reset, and I could reproduce it twice. Then I tried one more time to take screenshots and then for some reason colors are remaining, There is something a bit off.

tarman3, 2024-11-22:
Can not confirm in my system.
Try to disable **OpenTheme** and select FreeCAD Dark or Light theme.

luzpaz, 2024-11-22:
There is an issue where themes are hijacking or disabling colors. For example:  
* #17984

luzpaz, 2025-01-31:
@leoheck any luck recreating the issue ?

janjan, 2025-03-30:
I can reproduce this 100% of the time on my Linux and Windows systems:

- create a new part design object
- create a new sketch and draw a rectangle or circle or whatever
- pad the sketch
- right-click the PAD! (and not the body!) in the tree and select "random color" -> the object has a new color
- right-click the body and select "toggle transparency" -> the body switched back to its original color

If you randomize the color by clicking on the main object then it keeps its color if you toggle transparency

It only resets the color after toggling transparency if you set the color via right-clicking on the Pad object!

(I used the latest available weekly at the time of writing this comment: FreeCAD_weekly-builds-40971-conda-Linux-x86_64-py311)

janjan, 2025-04-05:
I did some digging in the code and I am pretty sure the problem is that Std_RandomColor assigns the color to the tip of the Part Design object and not onto the Part Design object itself. 

A similar issue existed/exists for toggling the transparency (#11353) and visibility (#11132) of Part Design objects

It should be quite easy to copy/paste/modify the fix for 11353 in Std_ToggleTransparency over to Std_RandomColor

yusufm, 2025-10-03:
If you set the color on a PartDesign feature (e.g., a Pad) but then run Toggle Transparency, the command redirects and toggles the Body’s Transparency. The Body’s ShapeAppearance color may still be the default gray. After toggling, the view can render with the Body’s appearance, so you see gray even though the Pad was set to blue.

The transparency change itself does not overwrite color; it likely changes which ViewProvider’s appearance is effectively used.

leoheck, 2025-10-28:
Another related issue is that if we set colors to faces and then export a step file, those colors do not get exported.

maxwxyz, 2025-11-01:
https://codeberg.org/xCAD/FreeCAD11/pulls/59/commits/f2c7396133b82b9ea5798efca289079c210515b8 is marked to fix this issue

closed 2025-11-03 by chennes