FreeCAD/FreeCAD #18076: [TNP] Fillet: Invalid edge link after changing the tip to a second pad

amreo, 2024-11-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Freecad fails to reassign the fillet edge to the new edge after a tip change with the error:
```
13:23:04  <PropertyLinks> PropertyLinks.cpp(447): Unnamed#Fillet.Base missing element reference Unnamed#Pad001 ;#5;:G;XTR;:H8fb:7,E.Edge1
13:23:05  <Exception> FeatureDressUp.cpp(209): Invalid edge link: ;#5;:G;XTR;:H8fb:7,E.Edge1
13:23:05  Fillet: Invalid edge link: ;#5;:G;XTR;:H8fb:7,E.Edge1
```

Steps to reproduce:
1) New document
2) Go to part design and create a new sketch
3) Click to the XY plane
4) Add a rectangle centered around the origin point
5) Close the sketch
6) Extrude the sketch of 20mm
7) Create a fillet of 5mm around edge1 (the bottom right corner on the XY plane)
8) Change tip(final entity) to the Pad
9) Create a sketch on face6 (the top face)
10) Create a external geometry(Skecther_external) copying Vertex2 (bottom right corner on the XY plane)
11) Create a rectangle from the copy of the vertex toward the center of the pad
12) Close the sketch
13) Pad the rectangle of 20mm

Freecad fails to update the fillet pointing to Pad001.Edge21 with the error:
```
18:03:41  <PropertyLinks> PropertyLinks.cpp(447): Unnamed1#Fillet.Base missing element reference Unnamed1#Pad001 ;#5;:G;XTR;:H11f1:7,E.Edge1
18:03:41  <Exception> FeatureDressUp.cpp(209): Invalid edge link: ;#5;:G;XTR;:H11f1:7,E.Edge1
18:03:41  Fillet: Invalid edge link: ;#5;:G;XTR;:H11f1:7,E.Edge1
18:03:41  <Exception> FeatureDressUp.cpp(209): Invalid edge link: ;#5;:G;XTR;:H11f1:7,E.Edge1
18:03:41  Fillet: Invalid edge link: ;#5;:G;XTR;:H11f1:7,E.Edge1
18:03:47  <Exception> FeatureDressUp.cpp(209): Invalid edge link: ;#5;:G;XTR;:H11f1:7,E.Edge1
18:03:47  Fillet: Invalid edge link: ;#5;:G;XTR;:H11f1:7,E.Edge1
```

### Full version info

```shell
OS: KDE Flatpak runtime (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * fasteners 0.5.25
  * 3DfindIT 1.2.0
  * freecad.gears 1.0.0
  * LCInterlocking
  * ModernUI (Disabled)
  * Beltrami 1.0.8-alpha
  * CurvedShapes 1.0.9
  * boltsfc 2022.11.5
  * ThreadProfile 1.92.0
  * sheetmetal 0.4.23
  * lattice2 1.0.0
  * Curves 0.6.42
  * Assembly4 0.50.6
  * Assembly3 0.12.2
  * A2plus 0.4.65
```

### Subproject(s) affected?

PartDesign

### Anything else?

[tnp_bug.FCStd.zip](https://github.com/user-attachments/files/17873700/tnp_bug.FCStd.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-11-22:
Freecad can not to reassign the fillet.
This chamfer breaking problem is still present

maxwxyz, 2024-11-23:
Reordering tip is basically like deleting a feature in between. Currently there is no automatic workaround but I hope there will be a fix implemented for 1.1. Closing this in favor of https://github.com/FreeCAD/FreeCAD/issues/14643

closed 2024-11-23 by maxwxyz