FreeCAD/FreeCAD #18078: Step Export to inch is creating a model in mm

vernzimm, 2024-11-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have Step file export unit set to inches. With newest nightly build it exports the geometry data (the actual numerical information) as millimeters, but still tags the file as inch, meaning the model scales up in size. Previous nightly build I was using was 38684.

I do see there was an update from Open CASCADE 7.7.2 to 7.8.1. I did take a look at their tracker as well, but don't see any mention of an issue like this.

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.39240 (Git) Conda
Build type: Release
Branch: main
Hash: f9ba94c70b139ece6ba5ffba47632d91b5c9c645
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * ProDarkThemePreferencePack 1.0.0
  * ThreadProfile 1.92.0
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-11-22:
Confirm with 1.0
Do not confirm with weekly-builds-39240

vernzimm, 2024-11-22:
No, I made the model from scratch from a sketch in 39240.

I do notice though that your OCC version for 39240 is 7.7.2?

tarman3, 2024-11-22:
I will try to download 1.0.0_appimage and check.

tarman3, 2024-11-22:
Ok, to be clear write my test.
- Open FreeCAD
- File > New
- Open workbench Part
- Create default Cube
- Select Cube
- File > Export > Step with colors (*.step, *.stp) > Units for export of STEP = Inch
- File > Import > All settings by default

Issue **Confirmed**
```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.31; sheetmetal 0.5.7; Silk 0.1.5; PieMenu 1.9.5 (Disabled); Curves 0.6.51
```

Issue **NOT** confirmed
```OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.31; sheetmetal 0.5.7; Silk 0.1.5; PieMenu 1.9.5 (Disabled); Curves 0.6.51
```

Issue **NOT** confirmed
```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39240 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: f9ba94c70b139ece6ba5ffba47632d91b5c9c645
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.31; sheetmetal 0.5.7; Silk 0.1.5; PieMenu 1.9.5 (Disabled); Curves 0.6.51
```

tarman3, 2024-11-22:
I look inside step files.

It seems that the versions _1.0.0.39109 (Git) Conda AppImage_ and  _1.0.0.39109 (Git) Conda AppImage_ ignore the Inch preference.
So it seems to be working fine.
But inside the files the parameters are indicated in mm, not inches.

vernzimm, 2024-11-23:
Note Open CASCADE (OCC) versions again. Issue confirmed with OCC 7.8.1, issue not confirmed with OCC 7.7.2.

tarman3, 2024-11-23:
Also confirmed in Linuxmint 21.3 with **OCC 7.8.1**
```
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39256 (Git)
Build type: Release
Branch: main
Hash: 6f23f01e509348a6755ad3c465a3d7ffd758ee03
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

cweickhmann, 2025-01-15:
I can confirm a similar issue with the current FreeCAD AppImage and OCC 7.7.2:

1. Create a Body and a Cylinder object with dimensions: R = 100 mm and H = 50 mm.
2.A.1. Export as STEP selecting Units to be Millimetres
2.A.2. Open STEP file in Gmsh and observe a radius of 100 and a height of 50. That's okay. But:
2.B.1. Export as STEP selecting Units to be Metres
2.B.2. Open STEP file in Gmsh and observe a radius of still 100 and a height of 50. That should be 100e-3 and 50e-3, respectively.

FreeCAD Version Details:
```
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

OCC Version according to library listing:
```
Open CASCADE Technology
Version: 7.7.2
```

Host OS/arch:
```
Distributor ID: Ubuntu
Description:    Ubuntu 24.04.1 LTS
Release:        24.04
Codename:       noble
Kernel: Linux 6.8.0-49-generic
Arch: x86_64
```

vernzimm, 2025-02-12:
@cweickhmann interesting. I personally would never have need to export in meters vs mm :)

I checked this with most recent nightly and the issue is still there, but I found the culprit on OCCT bugtracker...

[https://github.com/Open-Cascade-SAS/OCCT/issues/354](url)

It looks like they are shooting for 02/17/25 for release of 7.9.0. Can this be added as a goal for FreeCAD to implement new OCCT version soon when it is released?

cweickhmann, 2025-02-13:
Well, the use case is pretty straight forward: when you use FreeCAD to generate geometries for simulations, you want to know your units. :-) I don't care too much if it's mm or m, but it should be what's written on the package.

vernzimm, 2025-02-25:
OCCT 7.9.0 has been released, likely fixing this issue. Please upgrade when possible/feasible.

https://github.com/Open-Cascade-SAS/OCCT/releases/tag/V7_9_0

vernzimm, 2025-04-16:
They are actually close to releasing 8.0.0... but FreeCAD most recent build is still using bugged 7.8.1? Anyone? I have been using build  38684 since Sept 2024 because I need to export .step files in the correct proportions... Anyone?

[OCCT Releases](https://github.com/Open-Cascade-SAS/OCCT/releases)

vernzimm, 2025-04-16:
I'm not sure I follow, but maybe I'm misunderstanding. It seems to be complete and released by OCCT? I do understand it's not a drop-in upgrade, there are some changes that need to be made on FreeCAD's side...

oursland, 2025-04-16:
FreeCAD uses a lot of package dependencies, each of those must be made compatible with OCCT 7.9.0 before FreeCAD can use OCCT 7.9.0.  That has not yet been completed.

dxdesjardins, 2025-06-07:
On 1.0.1, when I export to inch, it creates a STEP with units of mm.  When I export to mm, it creates a STEP with units of inch.

vernzimm, 2025-06-19:
> On 1.0.1, when I export to inch, it creates a STEP with units of mm. When I export to mm, it creates a STEP with units of inch.

Not sure if something was changed since then. Still using the same version of OCCT, but with weekly build 42180 it reliably exports in mm, regardless of what unit you are using to model or have set to export. Just make sure step export is set to mm and model will be the right size.

FEA-eng, 2025-12-27:
Still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20251223 (Git shallow)
Build date: 2025/12/23 18:19:32
Build type: Release
Branch: (HEAD detached at 065fea4)
Hash: 065fea473f080691988bb11092759bf40f3c46fe
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

vernzimm, 2025-12-28:
To be fair... I have it set to mm export, which does work correctly, and it's not really clear why it should have a unit selection in step format anyway. Maybe just set it to mm default always and be done with it?

ipatch, 2025-12-31:
i build and install freecad locally using cmake + ninja from a CLI, and i pull-in / use the required deps built via homebrew. i believe right i only have occ v7.9.2 installed on my asahi linux box as that is the upstream homebrew-core version of occ. in theory i could install another version of occ to test with, but that would require some additional work.

cweickhmann, 2025-12-31:
> [Image de pistage][https://github.com/notifications/beacon/ADT5B5SIHKEJ4XQIYPOIJ2L4ERK47A5CNFSM6AAAAACQC34UJ2WGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTW4W7HFC.gif]

Hooray, that's good news. I'll check if I can build it like this too. Happy new year everyone, btw.

stefano2734, 2026-09-18:
Can be marked with label to 7.9.0 and higher.
Here important for users of occt 7.8.1 which is normal in 1.1.x and normal 26.3 dev versions wit occt 7.8.1.

luzpaz, 2026-09-21:
This is fixed upstream. Users will see the fix when #32431 is implemented

closed 2026-09-21 by luzpaz