FreeCAD/FreeCAD #18080: Wrong CMYK color code in documentation

porgand, 2024-11-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On logo guidelines: https://fpa.freecad.org/handbook/process/logo.html
it says about black: CMYK(0,0,16,1)
CMYK 0,0,16,1 is not black, should be CMYK(0,0,16,82)

### Full version info

```shell
On logo guidelines: https://fpa.freecad.org/handbook/process/logo.html
it says about black: Off Black, #212529, RGBA(33,37,41,1), CMYK(0,0,16,1), PMS BLACK C

CMYK 0,0,16,1 is not black (it is pale yellow)
consulting with guidelines PDF
https://fpa.freecad.org/images/logos/guidelines/guidelines.pdf
it should be CMYK(0,0,16,82)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-11-22:
cc @obelisk79 @yorikvanhavre

yorikvanhavre, 2024-11-25:
> cc @obelisk79 @yorikvanhavre

Looks valid to me! Indeed most off-black colors I could find on the net have a K values around 70-80. Probably a typo...

obelisk79, 2024-11-25:
according to conversion @ https://colocode.io

it looks like it should be CMYK(3,2,0,84)
[image]

closed 2024-12-19 by yorikvanhavre