FreeCAD/FreeCAD #18082: [PartDesign] Fillet does not round correctly (can't reproduce?)

TheRealOne78, 2024-11-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a problem in which I applied a fillet and one side was rounded while the other was chamfered, thus making a non-closed object.

Here's a video explaining it better:

https://github.com/user-attachments/assets/3bd7710a-489c-4691-83c2-fd58f193a481

Unfortunately, I have no idea how to reproduce it again.

### Full version info

```shell
OS: Gentoo Linux (Xsession/xcb)
Architecture: x86_64
Version: 1.1.0dev.Unknown
Build type: Release
Python 3.12.7, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * 3DfindIT 1.2.0
  * Help 1.0.3
  * ProDarkThemePreferencePack 1.0.0
  * Gridfinity 0.6.0
  * FreeCAD-Gridfinity-Workbench 0.6.0
```

### Subproject(s) affected?

PartDesign

### Anything else?

I can't really determine which commit it was, but the source code was fetched about 3 days ago: 
```>>> Messages generated by process 2821 on 2024-11-19 18:47:22 EET for package media-gfx/freecad-9999:```

I really can't reproduce this again, despite using the same version.

I heard that FreeCAD uses python scripts to generate each thing, so I figured I'd send the original file here, because I have no idea how to extract the python script in both cases, and then diff them:  [something.zip](https://github.com/user-attachments/files/17875965/something.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-11-22:
> Version: 1.1.0dev.Unknown

Why ?

TheRealOne78, 2024-11-22:
> > Version: 1.1.0dev.Unknown
> 
> Why ?

I suppose that's what happens when you are compiling it from git. It was compiled using a live package on gentoo: https://gitweb.gentoo.org/repo/gentoo.git/tree/media-gfx/freecad/freecad-9999.ebuild

TheRealOne78, 2025-09-06:
> If the issue is still relevant, let us know by adding a comment. Otherwise, we’ll close this issue automatically in 60 days.

I see that [this file](https://github.com/user-attachments/files/17875965/something.zip) that I've mentioned above still has the same issue on both FreeCAD 1.0.1 and FreeCAD weekly-2025.09.03

However I don't know how this can be reproduced. I hoped that someone would analyze the python code for the fillet in the file, or at least guide me how to do that since I'm having a hard time sourcing documentation on that matter.

TheRealOne78, 2025-11-14:
> Is it still the case that you can't reproduce it?

Yea.. besides the wall being slanted, and slightly at an angle, I have absolutely no idea what went wrong there. At the time I noticed that, I was able to ctrl+z to redo it, but if I deleted the fillet and created a new one, it would generate it correctly.

FEA-eng, 2026-10-17:
[This](https://github.com/FreeCAD/FreeCAD/issues/18082#issuecomment-5001699015) is still relevant

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261013 (Git shallow)
Build date: 2026/10/13 19:55:30
Build type: Release
Branch: (HEAD detached at 50daee777)
Hash: 50daee777321b5085accd1f57b57f1a7c1781cae
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 617.14/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```