FreeCAD/FreeCAD #18084: Draft Workbench widget issue

Danial1985, 2024-11-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Annotation scale widget and Snap widget doubled on status bar in Draft Workbench. Second copy not fully functional and could not turned off, it appears again after reboot.
[image]

### Full version info

```shell
OS: Windows 10 build 14393
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Ukrainian/Ukraine (uk_UA)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Reinforcement
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-23:
@Roy-043 FYI. maybe related to https://github.com/FreeCAD/FreeCAD/issues/16832 ?

Danial1985, 2024-11-23:
Yes. Looks like you are right. I also have empty lines in View->Toolbars. Its do could be  related to #16832
[image]

Roy-043, 2024-11-25:
Closing this as completed with #18064.

closed 2024-11-25 by Roy-043