FreeCAD/FreeCAD #18085: Segfault using the new assembly workbench

csreid, 2024-11-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Apologies if it's a dupe, but I didn't find anything. 

Trying out 1.0 (so good) and I whipped up a little two-wheeled robot. In the new assembly workbench, I tried to create a pair of revolute joints to align the wheels to a pair of holes in the chassis which were created using one "through all" pocket. The first one went fine, but I tried to flip the direction of the LCS for the second one and got an instant crash. Stacktrace is attached.

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * Assembly4 0.50.15
  * freecad.cross 1.0.0
```

### Subproject(s) affected?

Assembly

### Anything else?

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x75798a442520]
#1  0x7579672a5083 in Assembly::AssemblyObject::isPartGrounded(App::DocumentObject*) from /tmp/.mount_FreeCADolLHK/usr/lib/AssemblyApp.so+0xd3
#2  0x7579672c66a2 in Assembly::AssemblyObjectPy::isPartGrounded(_object*) from /tmp/.mount_FreeCADolLHK/usr/lib/AssemblyApp.so+0x62
#3  0x7579672c6724 in Assembly::AssemblyObjectPy::staticCallback_isPartGrounded(_object*, _object*) from /tmp/.mount_FreeCADolLHK/usr/lib/AssemblyApp.so+0x24
#4  /tmp/.mount_FreeCADolLHK/usr/bin/../lib/libpython3.11.so.1.0(+0x206068) [0x75798ce06068]
#5  /tmp/.mount_FreeCADolLHK/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x75798cde5b5c]
#6  /tmp/.mount_FreeCADolLHK/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x75798cdf1532]
#7  /tmp/.mount_FreeCADolLHK/usr/bin/../lib/libpython3.11.so.1.0(+0x235df4) [0x75798ce35df4]
#8  /tmp/.mount_FreeCADolLHK/usr/bin/../lib/libpython3.11.so.1.0(+0x2357b0) [0x75798ce357b0]
#9  0x75798c3d81a7 in PySide::SignalManager::callPythonMetaMethod(QMetaMethod const&, void**, _object*, bool) from /tmp/.mount_FreeCADolLHK/usr/bin/../lib/./libpyside2.cpython-311-x86_64-linux-gnu.so.5.15+0x97
#10  0x75798c3d85b5 in PySide::SignalManager::qt_metacall(QObject*, QMetaObject::Call, int, void**) from /tmp/.mount_FreeCADolLHK/usr/bin/../lib/./libpyside2.cpython-311-x86_64-linux-gnu.so.5.15+0x125
#11  /tmp/.mount_FreeCADolLHK/usr/bin/../lib/libQt5Core.so.5(+0x2eb649) [0x75798aaeb649]
#12  0x75798ba5b522 in QAbstractButton::clicked(bool) from /tmp/.mount_FreeCADolLHK/usr/bin/../lib/libQt5Widgets.so.5+0x32
#13  /tmp/.mount_FreeCADolLHK/usr/bin/../lib/libQt5Widgets.so.5(+0x25b73a) [0x75798ba5b73a]
#14  /tmp/.mount_FreeCADolLHK/usr/bin/../lib/libQt5Widgets.so.5(+0x25d1c8) [0x75798ba5d1c8]
#15  0x75798ba5d3d7 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /tmp/.mount_FreeCADolLHK/usr/bin/../lib/libQt5Widgets.so.5+0xe7
#16  0x75798bb4610a in QToolButton::mouseReleaseEvent(QMouseEvent*) from /tmp/.mount_FreeCADolLHK/usr/bin/../lib/libQt5Widgets.so.5+0xa
#17  0x75798b9aec08 in QWidget::event(QEvent*) from /tmp/.mount_FreeCADolLHK/usr/bin/../lib/libQt5Widgets.so.5+0x178
#18  0x75798b9703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCADolLHK/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#19  0x75798b978262 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCADolLHK/usr/bin/../lib/libQt5Widgets.so.5+0x13a2
#20  0x75798d769559 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCADolLHK/usr/bin/../lib/libFreeCADGui.so+0xc9
#21  0x75798aab6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCADolLHK/usr/bin/../lib/libQt5Core.so.5+0x108
#22  0x75798b9763eb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCADolLHK/usr/bin/../lib/libQt5Widgets.so.5+0x1ab
#23  /tmp/.mount_FreeCADolLHK/usr/bin/../lib/libQt5Widgets.so.5(+0x1c6e6f) [0x75798b9c6e6f]
#24  /tmp/.mount_FreeCADolLHK/usr/bin/../lib/libQt5Widgets.so.5(+0x1ca0c9) [0x75798b9ca0c9]
#25  0x75798b9703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCADolLHK/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#26  0x75798d769559 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCADolLHK/usr/bin/../lib/libFreeCADGui.so+0xc9
#27  0x75798aab6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCADolLHK/usr/bin/../lib/libQt5Core.so.5+0x108
#28  0x75798b1353ec in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCADolLHK/usr/bin/../lib/libQt5Gui.so.5+0x84c
#29  0x75798b10b0bc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCADolLHK/usr/bin/../lib/libQt5Gui.so.5+0xac
#30  /tmp/.mount_FreeCADolLHK/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x64bea) [0x7579850a2bea]
#31  /tmp/.mount_FreeCADolLHK/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x757988d10ad7]
#32  /tmp/.mount_FreeCADolLHK/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x757988d13ac7]
#33  /tmp/.mount_FreeCADolLHK/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x757988d140ce]
#34  0x75798ab0aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCADolLHK/usr/bin/../lib/libQt5Core.so.5+0x66
#35  0x75798aab570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCADolLHK/usr/bin/../lib/libQt5Core.so.5+0x11b
#36  0x75798aabd212 in QCoreApplication::exec() from /tmp/.mount_FreeCADolLHK/usr/bin/../lib/libQt5Core.so.5+0x82
#37  0x75798d6c8744 in Gui::Application::runApplication() from /tmp/.mount_FreeCADolLHK/usr/bin/../lib/libFreeCADGui.so+0x8a4
#38  /tmp/.mount_FreeCADolLHK/usr/bin/freecad(+0x78b7) [0x5e7d87af08b7]
#39  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x75798a429d90]
#40  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x75798a429e40]
#41  /tmp/.mount_FreeCADolLHK/usr/bin/freecad(+0x7c2a) [0x5e7d87af0c2a]
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

csreid, 2024-11-23:
(n.b.: Assembly 4 was installed from a previous version and I was definitely using the new Assembly workbench here)

maxwxyz, 2024-11-23:
Can you provide the file and also which joint was flipped?

csreid, 2024-11-23:
I had moved on already and made some other changes, but I tried to reproduce a minimal example. I can get the crash from here by:

1. Activating the Assembly workbench
2. Activating  "Assembly" in the model
3. Hitting "Create revolute joint"
4. Clicking the face on the un-joined cylinder that's nearest to the cube
5. Hitting "reverse the direction of the joint" in the Create Joint dialogue

[crash.zip](https://github.com/user-attachments/files/17882966/crash.zip)

maxwxyz, 2024-11-24:
can confirm, usually you have two references already selected when clicking on the reverse button.
[image]

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.29.1
  * Curves 0.6.51
  * dodo 1.0.1
  * fasteners 0.5.31
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.5.6
  * Silk 0.1.5

```

PaddleStroke, 2025-03-07:
I think this bug was fixed since. I cannot reproduce. Please check @maxwxyz and close if necessary

maxwxyz, 2025-03-07:
there is no issue in the latest weekly, it is solved.

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/133.772
Installed mods: 
  * Cables 0.1.3
  * CfdOF 1.29.8
  * CurvedShapes 1.0.13
  * Curves 0.6.58
  * dodo 1.0.1
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.1.2
  * parts_library
  * Rocket 4.1.1
  * sheetmetal 0.7.21
  * Silk 0.1.5
  * Telemetry 1.0.0

```

closed 2025-03-07 by maxwxyz