FreeCAD/FreeCAD #18086: Clipping Plane must be reestablished across documents

nmschulte, 2024-11-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using the Clipping Plane feature, it seems to only affect the document that was active when it was opened, and must be closed and re-opened in order to affect a different active document.

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (xcb)
Architecture: x86_64
Version: 1.1.0dev.39287 (Git)
Build type: Debug
Branch: main
Hash: b5a43eb2208b91feb76b2cc363e3956922c3ad75
Python 3.12.7, Qt 5.15.15, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * Curves 0.6.13
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * boltsfc 2022.11.5
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-11-23:
Confirmed

https://github.com/user-attachments/assets/f4113764-f1af-4b84-8f83-dd4f0d0c017f

```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.31; sheetmetal 0.5.7; Silk 0.1.5; PieMenu 1.9.5 (Disabled); Curves 0.6.51
```

nmschulte, 2024-11-23:
The code and UX say "Toggles clipping plane for active view."  Does this imply a set of clipping parameters per-view?  The way `Gui::StdCmdToggleClipPlane` interoperates with `Gui::Dialog::Clipping` sort of allows this, but needs to track a set of dialogs per view, and handle hiding and showing them as the active view / document changes.  Otherwise, these two will need to change; I'm not sure if there's another similar per-view state pattern that could be referenced/lifted.

maxwxyz, 2024-11-23:
I hope this will be resolved by a general new approach of clipping planes https://github.com/FreeCAD/FreeCAD/issues?q=is%3Aissue+is%3Aopen+label%3A"Topic%3A+Clipping+Planes"

FEA-eng, 2026-10-10:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20261001 (Git shallow)
Build date: 2026/10/01 12:05:00
Build type: Release
Branch: (HEAD detached at 99c5620c5)
Hash: 99c5620c5f739ca5c308849dda95ac468fec8b79
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
OpenGL version/vendor/renderer: 4.6.0 - Build 27.20.100.8854/Intel/Intel(R) UHD Graphics 620
```