FreeCAD/FreeCAD #18093: getGlobalPlacement() not correct for some Links

Rexbas, 2024-11-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

@PaddleStroke, It looks like the global placement function does not return the right orientation for links that have the _Link Transform_ property set to true.
[image]

I also managed to make a link (with Link_Transform = false) that returns the incorrect orientation.

This causes the _Align to selection_ tool and the _Measure_ tool to not work properly for these type of links. Maybe it also affects assembly but I don't know how.

I attached two files below, if you use the _Align to selection_ tool you can see that it does not properly align.

### Full version info

```shell
OS: Fedora Linux 41 (KDE Plasma) (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39287 (Git)
Build type: Release
Branch: main
Hash: b5a43eb2208b91feb76b2cc363e3956922c3ad75
Python 3.11.0, Qt 5.15.8, Coin 4.0.3, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.25
```

### Subproject(s) affected?

None

### Anything else?

[Links.zip](https://github.com/user-attachments/files/17879783/Links.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-23:
@kadet1090 FYI

PaddleStroke, 2025-09-03:
Still relevant I believe

Rexbas, 2025-09-03:
Yes still relevant, I am writing some unit tests for global placement