FreeCAD/FreeCAD #18096: [segfault] Freecad not exporting STEP

leoheck, 2024-11-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This is annoying... 

```
case/v0.2/v0.2.0-WIP …
➜ ~/AppImages/freecad.appimage 
FreeCAD 1.0.0, Libs: 1.0.0R39109 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

NOTICE: Addon "/home/lheck/.local/share/FreeCAD/Mod/OpenTheme.backup1732209044.7838953" disabled by presence of ADDON_DISABLED stopfile
OSHAutoDoc workbench (v0.2.3) loaded
DiffTools workbench 0.1.0 loaded
NOTICE: Addon "/home/lheck/.local/share/FreeCAD/Mod/OpenTheme.backup1732209044.7838953" disabled by presence of ADDON_DISABLED stopfile
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
<Import> ExportOCAF2.cpp(574): LCS_Origin set default color
<Import> ExportOCAF2.cpp(387): PIR_Sensor_0_2_0#Constraints has null shape
<Import> ExportOCAF2.cpp(387): PIR_Sensor_0_2_0#Variables has null shape
<Import> ExportOCAF2.cpp(387): PIR_Sensor_0_2_0#Configurations has null shape
<Import> ExportOCAF2.cpp(574): LCS_PIR_ORing_Base1 set default color
<Import> ExportOCAF2.cpp(574): 0395105009 set default color
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x7c95f1c45320]
#1  /tmp/.mount_freecaNldHFa/usr/lib/./././libTKXDESTEP.so.7.7(+0x89291) [0x7c95a70c0291]
#2  /tmp/.mount_freecaNldHFa/usr/lib/./././libTKXDESTEP.so.7.7(+0x8a078) [0x7c95a70c1078]
#3  0x7c95a70c29e1 in STEPCAFControl_Writer::writeColors(opencascade::handle<XSControl_WorkSession> const&, NCollection_Sequence<TDF_Label> const&) from /tmp/.mount_freecaNldHFa/usr/lib/./././libTKXDESTEP.so.7.7+0x9e1
#4  0x7c95a70dfdff in STEPCAFControl_Writer::transfer(STEPControl_Writer&, NCollection_Sequence<TDF_Label> const&, STEPControl_StepModelType, char const*, bool, Message_ProgressRange const&) from /tmp/.mount_freecaNldHFa/usr/lib/./././libTKXDESTEP.so.7.7+0x17ef
#5  0x7c95a70e026f in STEPCAFControl_Writer::Transfer(opencascade::handle<TDocStd_Document> const&, STEPControl_StepModelType, char const*, Message_ProgressRange const&) from /tmp/.mount_freecaNldHFa/usr/lib/./././libTKXDESTEP.so.7.7+0x20f
#6  0x7c95a7fb7117 in Import::WriterStep::write(opencascade::handle<TDocStd_Document>) const from /tmp/.mount_freecaNldHFa/usr/lib/././Import.so+0x137
#7  /tmp/.mount_freecaNldHFa/usr/lib/ImportGui.so(+0x33eaa) [0x7c95c3d2deaa]
#8  /tmp/.mount_freecaNldHFa/usr/lib/ImportGui.so(+0x2f32d) [0x7c95c3d2932d]
#9  /tmp/.mount_freecaNldHFa/usr/bin/../lib/libFreeCADBase.so(method_keyword_call_handler+0x374) [0x7c95f38dc774]
#10  /tmp/.mount_freecaNldHFa/usr/bin/../lib/libpython3.11.so.1.0(+0x20603d) [0x7c95f460603d]
#11  /tmp/.mount_freecaNldHFa/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x7c95f45e5b5c]
#12  /tmp/.mount_freecaNldHFa/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7c95f45f1532]
#13  /tmp/.mount_freecaNldHFa/usr/bin/../lib/libpython3.11.so.1.0(+0x2abd1d) [0x7c95f46abd1d]
#14  /tmp/.mount_freecaNldHFa/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7c95f46ab794]
#15  /tmp/.mount_freecaNldHFa/usr/bin/../lib/libpython3.11.so.1.0(+0x2cabf3) [0x7c95f46cabf3]
#16  /tmp/.mount_freecaNldHFa/usr/bin/../lib/libpython3.11.so.1.0(+0x2c669a) [0x7c95f46c669a]
#17  /tmp/.mount_freecaNldHFa/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x7c95f46ba9c9]
#18  0x7c95f394641c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_freecaNldHFa/usr/bin/../lib/libFreeCADBase.so+0x6c
#19  0x7c95f4fb6ba4 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_freecaNldHFa/usr/bin/../lib/libFreeCADGui.so+0x94
[... 53 lines cut ...]
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.5.6
  * InventorLoader 1.4.0
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.31
  * osh-autodoc-workbench 0.2.3
  * OpenTheme.backup1732209044.7838953 2024.9.1 (Disabled)
  * OpticsWorkbench 1.0.26
  * lattice2 1.0.0
  * kicadStepUpMod 11.4.1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-11-23:
This problem with all models?
If you will try to export simple cube is the same?

maxwxyz, 2024-11-23:
Could it be the same as: https://github.com/FreeCAD/FreeCAD/issues/18056 ?

leoheck, 2024-11-23:
Single Bodies are exporting fine. The problem is happening when exporting a Part with links/fasteners/groups.

leoheck, 2024-11-23:
> Could it be the same as: #18056 ?

Possibly, the last line of the crash on both reports seem equal. I am comparing quickly from my phone, but the crash log looks the same.

tarman3, 2024-11-24:
Create part with links/fasteners/groups, an assembly.
**Can not** reproduce the issue.

<details>
  <summary>FreeCAD 1.0.0.39109 (Git), Arch Linux (KDE/plasma/wayland)</summary>

```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.31; sheetmetal 0.5.7; Silk 0.1.5; PieMenu 1.9.5 (Disabled); Curves 0.6.51
```
</details>

<details>
  <summary>FreeCAD 1.1.0dev.39240 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39240 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: f9ba94c70b139ece6ba5ffba47632d91b5c9c645
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.31; sheetmetal 0.5.7; Silk 0.1.5; PieMenu 1.9.5 (Disabled); Curves 0.6.51
```
</details>

leoheck, 2024-11-24:
@tarman3 thanks for following up. 
Cc @maxwxyz 

I think I found the culprit.
Now I have extra time to analyze and explain things better.

I have an Assembly (Asm4).
And I am importing some parts that include the ones said before and also some Parts. 

One of the Parts is a Kicad board, imported from the Step file into an individual Part.
There is no problem with the Imported board.

However, I made another Part to put the Board inside. 
Also, inside this new Part, is a version of this Board that was made Solid with Part WB (using `Convert to Solid`).

This `Board (Solid)` that was inside the Linked Part, was breaking Freecad when exporting the Step of the Assembly.

Here is the minimum Object tree that segfaults:

[image]

Now, creating a simple file from scratch that mimics this does not result in a segfault. 

However, checking the original design, the Solid created with the Board, is not well-formed, see its issues here:

[image]

If I understand well... the Step Export breaks when the design has issues, but it should not break Freecad. This seems similar to the issue that was linked here.

Also (a second problem) when creating a Solid, Freecad is producing a Solid with problems without telling the user. And this was not alerted on the Report View.

leoheck, 2024-11-24:
Ah, forgot the most important part. Exporting only the problematic solid works without faulting FreeCad.

leoheck, 2024-11-24:
Honestly, this was my first time using the `Part >> Create a solid`. 

Using the `Part >> Create a copy >> Create a Simple Copy` as I always use... exporting the Assembly does not segfault Freecad. The produced object has a lot more errors, but the segfault does not happen in this case.

[image]

gemarcano, 2025-03-26:
I think I'm having the same issue. Here's my backtrace from FreeCAD:

```
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib64/libc.so.6(+0x41fd0) [0x7f0916e41fd0]
#1  /usr/lib64/opencascade/libTKDESTEP.so.7.8(+0x4557e1) [0x7f089d2557e1]
#2  /usr/lib64/opencascade/libTKDESTEP.so.7.8(+0x456d21) [0x7f089d256d21]
#3  0x7f089d258cf1 in STEPCAFControl_Writer::writeColors(opencascade::handle<XSControl_WorkSession> const&, NCollection_Sequence<TDF_Label> const&) from /usr/lib64/opencascade/libTKDESTEP.so.7.8+0x891
#4  0x7f089d27386f in STEPCAFControl_Writer::transfer(STEPControl_Writer&, NCollection_Sequence<TDF_Label> const&, STEPControl_StepModelType, char const*, bool, Message_ProgressRange const&) from /usr/lib64/opencascade/libTKDESTEP.so.7.8+0x14cf
#5  0x7f089d273f7c in STEPCAFControl_Writer::Transfer(opencascade::handle<TDocStd_Document> const&, StepData_ConfParameters const&, STEPControl_StepModelType, char const*, Message_ProgressRange const&) from /usr/lib64/opencascade/libTKDESTEP.so.7.8+0x49c
#6  0x7f089d2740f2 in STEPCAFControl_Writer::Transfer(opencascade::handle<TDocStd_Document> const&, STEPControl_StepModelType, char const*, Message_ProgressRange const&) from /usr/lib64/opencascade/libTKDESTEP.so.7.8+0xb2
#7  0x7f0892820316 in Import::WriterStep::write(opencascade::handle<TDocStd_Document>) const from /usr/lib64/freecad/lib64/Import.so+0x136
#8  0x7f0890ec9b0f in ImportGui::Module::exporter(Py::Tuple const&, Py::Dict const&) from /usr/lib64/freecad/lib64/ImportGui.so+0xa6f
#9  0x7f0890ec58af in Py::ExtensionModule<ImportGui::Module>::invoke_method_keyword(void*, Py::Tuple const&, Py::Dict const&) from /usr/lib64/freecad/lib64/ImportGui.so+0x3f
#10  /usr/lib64/freecad/lib64/libFreeCADBase.so(method_keyword_call_handler+0x35a) [0x7f09198fcfaa]
#11  /usr/lib64/libpython3.12.so.1.0(+0x17c609) [0x7f091917c609]
#12  /usr/lib64/libpython3.12.so.1.0(_PyObject_MakeTpCall+0x84) [0x7f09190df364]
#13  /usr/lib64/libpython3.12.so.1.0(_PyEval_EvalFrameDefault+0x6e30) [0x7f09192f3eb0]
#14  /usr/lib64/libpython3.12.so.1.0(PyEval_EvalCode+0xee) [0x7f09193003ce]
#15  /usr/lib64/libpython3.12.so.1.0(+0x300f66) [0x7f0919300f66]
#16  /usr/lib64/libpython3.12.so.1.0(+0x301066) [0x7f0919301066]
#17  /usr/lib64/libpython3.12.so.1.0(PyRun_StringFlags+0x6f) [0x7f091930115f]
#18  0x7f091996b25a in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /usr/lib64/freecad/lib64/libFreeCADBase.so+0x6a
#19  0x7f091ab6168a in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x8a
#20  0x7f091a9ecb3a in Gui::Application::exportTo(char const*, char const*, char const*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x56a
#21  0x7f091ab80a7a in StdCmdExport::activated(int) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x93a
#22  0x7f091ab67d6e in Gui::Command::_invoke(int, bool) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x23e
#23  0x7f091ab68246 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x126
#24  /usr/lib64/libQt6Core.so.6(+0x36c09f) [0x7f091776c09f]
#25  0x7f091824fc56 in QAction::triggered(bool) from /usr/lib64/libQt6Gui.so.6+0x46
#26  0x7f0918252c65 in QAction::activate(QAction::ActionEvent) from /usr/lib64/libQt6Gui.so.6+0xf5
#27  /usr/lib64/libQt6Widgets.so.6(+0x3cab5d) [0x7f09189cab5d]
#28  /usr/lib64/libQt6Widgets.so.6(+0x44fd12) [0x7f0918a4fd12]
#29  0x7f0918c17c33 in QWidget::event(QEvent*) from /usr/lib64/libQt6Widgets.so.6+0xb13
#30  0x7f0918c04da5 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib64/libQt6Widgets.so.6+0x85
#31  0x7f0918c05b25 in QApplication::notify(QObject*, QEvent*) from /usr/lib64/libQt6Widgets.so.6+0xcc5
#32  0x7f091aaf7da8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0xc8
#33  0x7f091778a320 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib64/libQt6Core.so.6+0x160
#34  0x7f0918c0f19a in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /usr/lib64/libQt6Widgets.so.6+0x1ea
#35  /usr/lib64/libQt6Widgets.so.6(+0x612bd6) [0x7f0918c12bd6]
#36  /usr/lib64/libQt6Widgets.so.6(+0x61926a) [0x7f0918c1926a]
#37  0x7f0918c04da5 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib64/libQt6Widgets.so.6+0x85
#38  0x7f091aaf7da8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0xc8
#39  0x7f091778a320 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib64/libQt6Core.so.6+0x160
#40  0x7f0917da9b57 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /usr/lib64/libQt6Gui.so.6+0x377
#41  0x7f0917e0fe5c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib64/libQt6Gui.so.6+0xac
#42  /usr/lib64/qt6/plugins/platforms/../../../libQt6XcbQpa.so.6(+0x6501e) [0x7f09118ef01e]
#43  /usr/lib64/libglib-2.0.so.0(+0x644e2) [0x7f0915af54e2]
#44  /usr/lib64/libglib-2.0.so.0(+0xd67f8) [0x7f0915b677f8]
#45  /usr/lib64/libglib-2.0.so.0(g_main_context_iteration+0x30) [0x7f0915af5db0]
#46  0x7f09174cfa73 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib64/libQt6Core.so.6+0x73
[... cut ...]
I'm attaching the project file that's causing the crash. This was something I sketched out and made pre 1.0, and was trying to change the color and export to step and wrl for Kicad. To trigger the crash, all I do is select the part, go to File->Export, select `Step with color` file format, and just keep accepting defaults. It crashes before the file is written out to disk.

For reference, I on Gentoo using FreeCAD 1.0.0-r3, and opencascade-7.8.1.

I can probably help with debugging FreeCad or opencascade, if this can't be reproduced.

[controller_port_female.zip](https://github.com/user-attachments/files/19476298/controller_port_female.zip)

gemarcano, 2025-03-27:
Doing a little more poking around, there seems to be an issue in setting that override style. It really looks like it ought to be updated in a call to `getStyledItem(theShape, theShTool, theStyles, theOverride, theMapCompMDGPR);` in line 1220 (opencascade 7.8.1), but there's a check in there that's causing an early return (`if (aTopLevSh.IsNull() || !theMapCompMDGPR.IsBound(aTopLevSh))`). I don't know what MDGPR is, and why it's not bound at this point.

Just to sanity check, I downloaded the `FreeCAD_1.0.0-conda-Linux-x86_64-py311.AppImage` program, and ran it, and it also crashed just like my version. Looks like the appimage opencascade version is older than mine, at 7.7.2.

As another sanity check, I made a new project with just a part and a cube, and changed the colors, and it exported fine... so there may be something my project is doing that's different than a simple part and cube...

gemarcano, 2025-03-27:
Just found out with my project that if I don't export invisible objects the export succeeds! So it may be something related to invisible objects.

luzpaz, 2025-07-28:
Needs diagnosis

FEA-eng, 2025-09-27:
> I'm attaching the project file that's causing the crash. This was something I sketched out and made pre 1.0, and was trying to change the color and export to step and wrl for Kicad. To trigger the [...]

Maybe this is OS-specific because I don't get a crash with this file on Windows.

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

gemarcano, 2026-01-06:
And I also said:

> However, I cannot reproduce it with the 2025-12-31 weekly build:

So at least on my end it looks like it's been fixed? Can you reproduce it using their appimage from https://github.com/FreeCAD/FreeCAD/releases/tag/weekly-2025.12.31 ?