FreeCAD/FreeCAD #18097: Using What's this? function crashes FreeCAD when PieMenu is installed

hasecilu, 2024-11-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

when the option to open the Wiki page is either in a FreeCAD tab or a window.

I was able to isolate the problem on PieMenu by resetting my `user.cfg` file because with all my old settings the crash existed even without any addon.

I don't see anything obvious on PieMenuso I think something could be bad on FreeCAD.

[image]

I can not properly do a `git bisect` on PieMenu because of an older Qt6 crash. Also can not test on Qt5 because Arch Linux packages Qt6 version and AppImages with Qt5 says: "Could not find QtWebEngineProcess" so I have no way to know which commit introduced the problem.

The workaround for this is to use the web browser option for the Help module.

Other users have confirmed the problem on Linux with Qt5 builds.

### backtrace of SIGSEGV on Qt6 build

<details>
  <summary><b>Click here to show/hide</b></summary>

```gdb
Thread 1 "FreeCAD" received signal SIGSEGV, Segmentation fault.
0x00007ffff25e2754 in ?? () from /usr/lib/libpyside6.cpython-312-x86_64-linux-gnu.so.6.8
(gdb) bt
#0  0x00007ffff25e2754 in ??? () at /usr/lib/libpyside6.cpython-312-x86_64-linux-gnu.so.6.8
#1  0x00007ffff25e4a28 in PySide::getWrapperForQObject(QObject*, _typeobject*) () at /usr/lib/libpyside6.cpython-312-x86_64-linux-gnu.so.6.8
#2  0x00007fffb301760d in ??? () at /usr/lib/python3.12/site-packages/PySide6/QtWidgets.cpython-312-x86_64-linux-gnu.so
#3  0x00007fffef1553d8 in QCoreApplicationPrivate::sendThroughApplicationEventFilters
    (this=this@entry=0x555555aaf450, receiver=receiver@entry=0x55555ef97c30, event=event@entry=0x7fffffff6e60) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qcoreapplication.cpp:1284
#4  0x00007ffff00fe35a in QApplicationPrivate::notify_helper (this=0x555555aaf450, receiver=0x55555ef97c30, e=0x7fffffff6e60)
    at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qapplication.cpp:3265
#5  0x00007ffff4f95e20 in Gui::GUIApplication::notify (this=0x7fffffffc650, receiver=0x55555ef97c30, event=0x7fffffff6e60) at /home/uli/git/FreeCAD/FreeCAD/src/Gui/GuiApplication.cpp:93
#6  0x00007fffef1585a8 in QCoreApplication::notifyInternal2 (receiver=0x55555ef97c30, event=0x7fffffff6e60) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qcoreapplication.cpp:1165
#7  0x00007fffef1acbb4 in QCoreApplication::sendEvent (receiver=0x55555ef97c30, event=0x7fffffff6e60) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qcoreapplication.cpp:1609
#8  QObject::doSetProperty (this=0x55555ef97c30, name=<optimized out>, lvalue=<optimized out>, rvalue=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qobject.cpp:4307
#9  0x00007ffff25e4acc in PySide::getWrapperForQObject(QObject*, _typeobject*) () at /usr/lib/libpyside6.cpython-312-x86_64-linux-gnu.so.6.8
#10 0x00007fffb301760d in ??? () at /usr/lib/python3.12/site-packages/PySide6/QtWidgets.cpython-312-x86_64-linux-gnu.so
#11 0x00007fffef1553d8 in QCoreApplicationPrivate::sendThroughApplicationEventFilters
    (this=this@entry=0x555555aaf450, receiver=receiver@entry=0x55555ef97c30, event=event@entry=0x7fffffff7440) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qcoreapplication.cpp:1284
#12 0x00007ffff00fe35a in QApplicationPrivate::notify_helper (this=0x555555aaf450, receiver=0x55555ef97c30, e=0x7fffffff7440)
    at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qapplication.cpp:3265
#13 0x00007ffff4f95e20 in Gui::GUIApplication::notify (this=0x7fffffffc650, receiver=0x55555ef97c30, event=0x7fffffff7440) at /home/uli/git/FreeCAD/FreeCAD/src/Gui/GuiApplication.cpp:93
#14 0x00007fffef1585a8 in QCoreApplication::notifyInternal2 (receiver=0x55555ef97c30, event=0x7fffffff7440) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qcoreapplication.cpp:1165
#15 0x00007fffef1585ed in QCoreApplication::sendEvent (receiver=<optimized out>, event=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qcoreapplication.cpp:1609
#16 0x00007fff53b5a9de in QQuickDeliveryAgentPrivate::setFocusInScope (this=<optimized out>, scope=<optimized out>, item=<optimized out>, reason=<optimized out>, options=...)
    at /usr/src/debug/qt6-declarative/qtdeclarative/src/quick/util/qquickdeliveryagent.cpp:508
#17 0x00007fff539a7c1d in QQuickItem::setFocus (this=0x55555efbfeb0, focus=<optimized out>, reason=Qt::OtherFocusReason) at /usr/include/qt6/QtCore/qflags.h:73
#18 0x00007fff53a5823c in QQuickWindow::focusInEvent (this=<optimized out>, ev=<optimized out>) at /usr/src/debug/qt6-declarative/qtdeclarative/src/quick/items/qquickwindow.cpp:239
--Type <RET> for more, q to quit, c to continue without paging--
#19 0x00007ffff01510c9 in QWidget::event (this=0x55555efb6a60, event=0x7fffffff7b10) at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qwidget.cpp:9086
#20 0x00007ffff00fe31a in QApplicationPrivate::notify_helper (this=<optimized out>, receiver=0x55555efb6a60, e=0x7fffffff7b10)
    at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qapplication.cpp:3294
#21 0x00007ffff4f95e20 in Gui::GUIApplication::notify (this=0x7fffffffc650, receiver=0x55555efb6a60, event=0x7fffffff7b10) at /home/uli/git/FreeCAD/FreeCAD/src/Gui/GuiApplication.cpp:93
#22 0x00007fffef1585a8 in QCoreApplication::notifyInternal2 (receiver=0x55555efb6a60, event=0x7fffffff7b10) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qcoreapplication.cpp:1165
#23 0x00007fffef1585ed in QCoreApplication::sendEvent (receiver=<optimized out>, event=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qcoreapplication.cpp:1609
#24 0x00007ffff010b4c8 in QApplicationPrivate::setFocusWidget (focus=0x55555efb6a60, reason=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qapplication.cpp:1543
#25 0x00007ffff015fc76 in QWidget::setFocus (this=0x55555efb6a60, reason=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qwidget.cpp:6580
#26 0x00007ffff02d41d3 in QMdiSubWindowPrivate::setActive (this=this@entry=0x55555eef30f0, activate=activate@entry=true, changeFocus=changeFocus@entry=true)
    at /usr/src/debug/qt6-base/qtbase/src/widgets/widgets/qmdisubwindow.cpp:1451
#27 0x00007ffff02d762a in QMdiSubWindow::eventFilter (this=0x55555efc4410, object=0x55555eca0ac0, event=0x7fffffff8150)
    at /usr/src/debug/qt6-base/qtbase/src/widgets/widgets/qmdisubwindow.cpp:2679
#28 0x00007fffef155508 in QCoreApplicationPrivate::sendThroughObjectEventFilters (receiver=receiver@entry=0x55555eca0ac0, event=event@entry=0x7fffffff8150)
    at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qcoreapplication.cpp:1304
#29 0x00007ffff00fe30a in QApplicationPrivate::notify_helper (this=<optimized out>, receiver=0x55555eca0ac0, e=0x7fffffff8150)
    at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qapplication.cpp:3288
[... 137 lines cut ...]
```shell
OS: Arch Linux (XFCE/openbox/xcb)
Architecture: x86_64
Version: 1.1.0dev.36460 +2833 (Git)
Build type: Release
Branch: test/prs
Hash: 75283a56a9982657b98c248c33134d78aa141347
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: Spanish/Mexico (es_MX)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * PieMenu 1.10.0
```

### Subproject(s) affected?

Project Tools & Websites

### Anything else?

This is the issue I made on PieMenu repo but probably is something on FreeCAD, so I place it here also: https://github.com/Grubuntu/PieMenu/issues/114

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MiqCG, 2024-11-23:
I can reproduce the crash with a fresh "install" (deleted .cache .config .local/share) with appimages 39287, 39240 and 39175 revisions when "In a FreeCAD tab" is enabled. 

No FreeCAD installed in the system, no old configurations (I think). No Pie Menu installed in FreeCAD.

Maybe related to https://github.com/FreeCAD/FreeCAD/issues/14922#issuecomment-2203019526

OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.39287 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b5a43eb2208b91feb76b2cc363e3956922c3ad75
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

adrianinsaval, 2024-11-24:
Note that webengine is not included in the appimage so opening a tab in freecad will not work either way

tarman3, 2024-11-24:
Reproduced with
<details>
  <summary>FreeCAD 1.0.0.39109 Full version info</summary>

```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.31; sheetmetal 0.5.7; Silk 0.1.5; PieMenu 1.9.5 (Disabled); Curves 0.6.51
```
</details>

---

**Not** reproduced with
<details>
  <summary>FreeCAD 1.1.0dev.39240 (Git) Conda AppImage</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39240 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: f9ba94c70b139ece6ba5ffba47632d91b5c9c645
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.31; sheetmetal 0.5.7; Silk 0.1.5; PieMenu 1.9.5 (Disabled); Curves 0.6.51
```
</details>

tarman3, 2024-11-24:
I'm sorry. Perhaps I'm confused.

If backtrace is console output:

After start FreeCAD
```
[user@arch app]$ QT_QPA_PLATFORM=xcb ./FreeCAD_weekly-builds-39240-conda-Linux-x86_64-py311.AppImage
FreeCAD 1.1.0, Libs: 1.1.0devR39240 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
```
Crash and console output after used function **What`s This?**
```
Could not find QtWebEngineProcess
/tmp/.mount_FreeCALMOBgo/AppRun: строка 31: 33495 Аварийный останов         (образ памяти сброшен на диск) "${MAIN}" "$@"
```

tarman3, 2024-11-24:
https://github.com/user-attachments/assets/751a6313-be19-4f8d-9cc9-0c66973336f5

tarman3, 2024-11-24:
I think the problem in AppImage.

I checked version 1.0.0 39109 from Archlinux repository and AppImage package 
Reproduced only with AppImage package.

MiqCG, 2024-11-24:
> @MiqCG, could you upload the backtrace?

```
$ ./FreeCAD_weekly-builds-39287-conda-Linux-x86_64-py311.AppImage 
FreeCAD 1.1.0, Libs: 1.1.0devR39287 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Could not find QtWebEngineProcess
/tmp/.mount_FreeCADlEGfk/AppRun línia 31:  9517 Avortat                 "${MAIN}" "$@"
```
and crash

Roy-043, 2024-11-24:
I can reproduce on Windows, although there is no crash there. The following warning appears in the Report view:

> QWindowsWindow::setMouseGrabEnabled: Not setting mouse grab for invisible window QWidgetWindow/'QWhatsThatClassWindow'

```
OS: Windows 8 build 9600
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods:
```

xtemp09, 2024-11-24:
So, one crash is a packaging issue, another one is related to _PieMenu_.

hasecilu, 2024-11-24:
Yes but I'm not completely sure that the real problem is on PieMenu because some users have reproduce without installing PieMenu, in my case yes, when I disable it the command works again.