FreeCAD/FreeCAD #18101: The refactored* postprocessors do not handle rotary axis parameters correctly

LarryWoestman, 2024-11-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The refactored* postprocessors treat the rotary axis parameters (A, B, & C) the same as the linear axis parameters (X, Y, & Z).  In particular, rotary axis parameter values (which should always be in degrees) are converted "from mm to inches" when the "--inches" argument is passed to the postprocessors.  No conversion is necessary or desired.
More tests are needed to check the rotary axis parameters under more circumstances.

Please assign this issue to me.

### Full version info

```shell
All versions of FreeCAD for at least the last year up to this point.
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

LarryWoestman, 2024-11-24:
PR @18103 will fix this issue.

DerAndere1, 2024-11-26:
Do you know if feedrate is handled correctly when rotary axes are involved? I develop CNC firmware and I know that many projects struggle with the feedrate interpretation. The best definition of the F parameter whith rotational axes I could find is [this](https://www.linuxcnc.org/docs/html/gcode/machining-center.html#sub:feed-rate) :

> The rate at which the controlled point moves is nominally a steady rate which may be set by the user. In the Interpreter, the feed rate is interpreted as follows (unless inverse time feed or feed per revolution modes are being used, in which case see section [G93-G94-G95-Mode](https://www.linuxcnc.org/docs/html/gcode/g-code.html#gcode:g93-g94-g95)).

>  If any of XYZ are moving, F is in units per minute in the XYZ cartesian system, and all other axes (ABCUVW) move so as to start and stop in coordinated fashion.

> Otherwise, if any of UVW are moving, F is in units per minute in the UVW cartesian system, and all other axes (ABC) move so as to start and stop in coordinated fashion.

> Otherwise, the move is pure rotary motion and the F word is in rotary units in the ABC pseudo-cartesian system.

Reference:

Kramer, T. , Proctor, F. and Messina, E. (2000), The NIST RS274NGC Interpreter - Version 3, NIST Interagency/Internal Report (NISTIR), National Institute of Standards and Technology, Gaithersburg, MD, [online], https://tsapps.nist.gov/publication/get_pdf.cfm?pub_id=823374 (Accessed November 26, 2024)

LarryWoestman, 2024-11-28:
I replied by email, but my response doesn't seem to have gotten here.

I don't know if feedrate is handled correctly when rotary axes are involved.  I am not aware of any tests for this area for any of the postprocessors.  I agree with the definitions you provided.

Once the tests for the issue that these comments are in are done, I would like to figure out some tests for feedrate for the refactored* postprocessors.  At a minimum, the postprocessors should "do the right thing" with whatever feed rate they get from the rest of the FreeCAD code.

closed 2024-12-13 by chennes