FreeCAD/FreeCAD #18104: Linux Mint 22 /Ubuntu 22, PySide6 problem

iplayfast, 2024-11-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I've just upgraded my system and can no longer build freecad. :( Actually it builds but when it's run 
```
./bin/FreeCAD
FreeCAD 1.1.0, Libs: 1.1.0devR39288 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

During initialization the error "No module named 'PySide6'" occurred in /home/chris/github/FreeCAD/build/Mod/Tux/InitGui.py
Please look into the log file for further information
During initialization the error "No module named 'PySide6'" occurred in /home/chris/github/FreeCAD/build/Mod/BIM/InitGui.py
Please look into the log file for further information
During initialization the error "No module named 'PySide6'" occurred in /home/chris/github/FreeCAD/build/Mod/Assembly/InitGui.py
Please look into the log file for further information
During initialization the error "No module named 'PySide6'" occurred in /home/chris/github/FreeCAD/build/Mod/AddonManager/InitGui.py
Please look into the log file for further information
During initialization the error "No module named 'PySide6'" occurred in /home/chris/github/FreeCAD/build/Mod/Measure/InitGui.py
Please look into the log file for further information
During initialization the error "No module named 'PySide6'" occurred in /home/chris/github/FreeCAD/build/Mod/Help/InitGui.py
Please look into the log file for further information
During initialization the error "No module named 'PySide6'" occurred in /home/chris/github/FreeCAD/build/Mod/Draft/InitGui.py
Please look into the log file for further information
Wizard shaft module cannot be loaded
Involute gear module cannot be loaded
Unknown command 'Std_AddonMgr'
Unknown command 'PartDesign_Sprocket'

```

I believe all this is because the PySide6 is not supported in Ubuntu 22. When trying to install with pip I get 
```
pip3 install pyside6
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.12/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.

```
and when trying to install using apt, it's just not available. 

Any suggestions? (and I really don't want a different virtual environtment for every app I run). 

### Full version info

```shell
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39288 (Git)
Build type: Unknown
Branch: main
Hash: 45134cef165ecc11aca69f81673e7b69421cebf1
Python 3.12.3, Qt 6.4.2, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * freecad.gears 1.3.0
  * Curves 0.6.51
  * CurvedShapes 1.0.13
```

### Subproject(s) affected?

None

### Anything else?

FreeCAD
List of Open Source components

FreeCAD uses the following Open Source Components.
[Boost](https://github.com/FreeCAD/FreeCAD/issues/new?assignees=&labels=Status%3A+Needs+triage%2CStatus%3A+Needs+confirmation&projects=&template=PROBLEM_REPORT.yml#_TocBoost)
[Coin3D](https://github.com/FreeCAD/FreeCAD/issues/new?assignees=&labels=Status%3A+Needs+triage%2CStatus%3A+Needs+confirmation&projects=&template=PROBLEM_REPORT.yml#_TocCoin3D)
[Eigen](https://github.com/FreeCAD/FreeCAD/issues/new?assignees=&labels=Status%3A+Needs+triage%2CStatus%3A+Needs+confirmation&projects=&template=PROBLEM_REPORT.yml#_TocEigen)
[FreeType](https://github.com/FreeCAD/FreeCAD/issues/new?assignees=&labels=Status%3A+Needs+triage%2CStatus%3A+Needs+confirmation&projects=&template=PROBLEM_REPORT.yml#_TocFreeType)
[KDL](https://github.com/FreeCAD/FreeCAD/issues/new?assignees=&labels=Status%3A+Needs+triage%2CStatus%3A+Needs+confirmation&projects=&template=PROBLEM_REPORT.yml#_TocKDL)
[libarea](https://github.com/FreeCAD/FreeCAD/issues/new?assignees=&labels=Status%3A+Needs+triage%2CStatus%3A+Needs+confirmation&projects=&template=PROBLEM_REPORT.yml#_TocLibArea)
[Open CASCADE Technology](https://github.com/FreeCAD/FreeCAD/issues/new?assignees=&labels=Status%3A+Needs+triage%2CStatus%3A+Needs+confirmation&projects=&template=PROBLEM_REPORT.yml#_TocOCCT)
[Point Cloud Library](https://github.com/FreeCAD/FreeCAD/issues/new?assignees=&labels=Status%3A+Needs+triage%2CStatus%3A+Needs+confirmation&projects=&template=PROBLEM_REPORT.yml#_TocPcl)
[PyCXX](https://github.com/FreeCAD/FreeCAD/issues/new?assignees=&labels=Status%3A+Needs+triage%2CStatus%3A+Needs+confirmation&projects=&template=PROBLEM_REPORT.yml#_TocPyCXX)
[Python](https://github.com/FreeCAD/FreeCAD/issues/new?assignees=&labels=Status%3A+Needs+triage%2CStatus%3A+Needs+confirmation&projects=&template=PROBLEM_REPORT.yml#_TocPython)
[Shiboken/PySide](https://github.com/FreeCAD/FreeCAD/issues/new?assignees=&labels=Status%3A+Needs+triage%2CStatus%3A+Needs+confirmation&projects=&template=PROBLEM_REPORT.yml#_TocPySide)
[Qt](https://github.com/FreeCAD/FreeCAD/issues/new?assignees=&labels=Status%3A+Needs+triage%2CStatus%3A+Needs+confirmation&projects=&template=PROBLEM_REPORT.yml#_TocQt)
[Salome SMESH](https://github.com/FreeCAD/FreeCAD/issues/new?assignees=&labels=Status%3A+Needs+triage%2CStatus%3A+Needs+confirmation&projects=&template=PROBLEM_REPORT.yml#_TocSalomeSMESH)
[Visualization Toolkit](https://github.com/FreeCAD/FreeCAD/issues/new?assignees=&labels=Status%3A+Needs+triage%2CStatus%3A+Needs+confirmation&projects=&template=PROBLEM_REPORT.yml#_TocVtk)
[Xerces-C](https://github.com/FreeCAD/FreeCAD/issues/new?assignees=&labels=Status%3A+Needs+triage%2CStatus%3A+Needs+confirmation&projects=&template=PROBLEM_REPORT.yml#_TocXercesC)
[Zipios++](https://github.com/FreeCAD/FreeCAD/issues/new?assignees=&labels=Status%3A+Needs+triage%2CStatus%3A+Needs+confirmation&projects=&template=PROBLEM_REPORT.yml#_TocZipios)
[zlib](https://github.com/FreeCAD/FreeCAD/issues/new?assignees=&labels=Status%3A+Needs+triage%2CStatus%3A+Needs+confirmation&projects=&template=PROBLEM_REPORT.yml#_TocZlib) 
Boost
Version: 1.83.0
Web site: https://www.boost.org/
License: Boost Software License: https://www.boost.org/users/license.html 

Coin3D
Version: 4.0.2
Web site: https://coin3d.github.io/
License: BSD 3-Clause: https://github.com/coin3d/coin/blob/master/COPYING 

Eigen
Version: 3.4.0
Web site: https://eigen.tuxfamily.org/
License: MPL version 2.0: https://www.mozilla.org/en-US/MPL/2.0/ 

FreeType
Version: 2.13.2
Web site: https://www.freetype.org/
License: FreeType License: https://gitlab.freedesktop.org/freetype/freetype/-/blob/master/docs/FTL.TXT 

KDL
Web site: https://www.orocos.org/kdl.html
License: LGPL version 2.1: [LGPL v2.1](https://github.com/FreeCAD/FreeCAD/issues/new?assignees=&labels=Status%3A+Needs+triage%2CStatus%3A+Needs+confirmation&projects=&template=PROBLEM_REPORT.yml#_LGPLv21) 

libarea
Web site: https://github.com/danielfalck/libarea/
License: BSD-3-clause: https://github.com/danielfalck/libarea/blob/master/License.txt 

Ondsel Solver
Version: 1.0.1
Web site: https://github.com/Ondsel-Development/OndselSolver
License: LGPL version 2.1: [LGPL v2.1](https://github.com/FreeCAD/FreeCAD/issues/new?assignees=&labels=Status%3A+Needs+triage%2CStatus%3A+Needs+confirmation&projects=&template=PROBLEM_REPORT.yml#_LGPLv21) 

Open CASCADE Technology
Version: 7.8.1
Web site: https://www.opencascade.com/open-cascade-technology/
License: LGPL version 2.1 with additional exception:https://dev.opencascade.org/resources/licensing 

Point Cloud Library
Version: 
Web site: https://www.pointclouds.org/
License: BSD-3-clausehttps://github.com/PointCloudLibrary/pcl/blob/master/LICENSE.txt 

PyCXX
Version: 7.1.9
Web site: http://cxx.sourceforge.net/
License: BSD-3-clause
                           Copyright (c) 1998 - 2007
                  The Regents of the University of California
             Produced at the Lawrence Livermore National Laboratory
             Written by Geoff Furnish, Paul F. Dubois, Barry A. Scott
                               UCRL-CODE-227018
                             All rights reserved.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mnesarco, 2024-12-13:
I have a similar problem. I compile in nix shell, everything compiles. but when run, no PySide6 or pivy available, but they are there.

Config:
```
  ==============
     System
   ==============

-- bindir:                      bin
-- BLAS:                        -undefined-
-- CMAKE_VERSION:               3.30.5
-- Compiler:                    /nix/store/4apajimszc47rxwcpvc3g3rj2icinl71-gcc-wrapper-13.3.0/bin/g++ (13.3.0)
-- datadir:                     share
-- docdir:                      share/doc/FreeCAD
-- includedir:                  include
-- libdir:                      lib
-- prefix:                      /var/empty/local
-- Python:                      3.11.10 [/nix/store/nmqxyr00in2arwrq5qd1qipsanz1yrn5-python3-3.11.10/bin/python3.11] Suffix: [.abi3]

   ==============
     Config
   ==============

-- BUILD_ADDONMGR:              ON
-- BUILD_ASSEMBLY:              ON
-- BUILD_BIM:                   ON
-- BUILD_CAM:                   ON
-- BUILD_CLOUD:                 OFF 
-- BUILD_DRAFT:                 ON
-- BUILD_DRAWING:               OFF 
-- BUILD_FEM:                   ON
-- BUILD_GUI:                   ON
-- BUILD_HELP:                  ON
-- BUILD_IDF:                   ON
-- BUILD_IMPORT:                ON
-- BUILD_INSPECTION:            ON
-- BUILD_JTREADER:              OFF 
-- BUILD_MATERIAL:              ON
-- BUILD_MEASURE:               ON
-- BUILD_MESH:                  ON
-- BUILD_MESH_PART:             ON
-- BUILD_OPENSCAD:              ON
-- BUILD_PART:                  ON
-- BUILD_PART_DESIGN:           ON
-- BUILD_PLOT:                  ON
-- BUILD_POINTS:                ON
-- BUILD_REVERSEENGINEERING:    ON
-- BUILD_ROBOT:                 ON
-- BUILD_SANDBOX:               OFF 
-- BUILD_SHOW:                  ON
-- BUILD_SKETCHER:              ON
-- BUILD_SMESH:                 ON
-- BUILD_SPREADSHEET:           ON
-- BUILD_START:                 ON
-- BUILD_SURFACE:               ON
-- BUILD_TECHDRAW:              ON
-- BUILD_TEST:                  ON
-- BUILD_TUX:                   ON
-- BUILD_VR:                    OFF 
-- BUILD_WEB:                   ON
-- CMAKE_BUILD_TYPE:            OFF 
-- CMAKE_CXX_FLAGS:             -Wall -Wextra -Wno-write-strings 
-- CMAKE_CXX_STANDARD:          17
-- CMAKE_INSTALL_PREFIX:        /var/empty/local
-- CMAKE_PREFIX_PATH:           OFF 
-- ENABLE_DEVELOPER_TESTS:      ON
-- FREECAD_CREATE_MAC_APP:      -undefined-
-- FREECAD_LIBPACK_USE:         -undefined-
-- FREECAD_QT_VERSION:          Auto
-- FREECAD_USE_EXTERNAL_KDL:    OFF 
[... 130 lines cut ...]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/mnesarco/.local/share/FreeCAD/Mod/MnesarcoUtils/freecad/mnesarco/init_gui.py", line 22, in <module>
    from freecad.mnesarco.scripts.hook import init_gui_scripts
  File "/home/mnesarco/.local/share/FreeCAD/Mod/MnesarcoUtils/freecad/mnesarco/scripts/hook.py", line 22, in <module>
    from freecad.mnesarco.utils import toolbars
  File "/home/mnesarco/.local/share/FreeCAD/Mod/MnesarcoUtils/freecad/mnesarco/utils/toolbars.py", line 21, in <module>
    from freecad.mnesarco.utils import preferences
  File "/home/mnesarco/.local/share/FreeCAD/Mod/MnesarcoUtils/freecad/mnesarco/utils/preferences.py", line 22, in <module>
    from freecad.mnesarco.utils.extension import log_err
  File "/home/mnesarco/.local/share/FreeCAD/Mod/MnesarcoUtils/freecad/mnesarco/utils/extension.py", line 22, in <module>
    from pivy import coin
ModuleNotFoundError: No module named 'pivy'
--------------------------------------------------------------------------------
Wizard shaft module cannot be loaded
Involute gear module cannot be loaded
Unknown command 'Std_AddonMgr'
Unknown command 'PartDesign_Sprocket'

```

mnesarco, 2024-12-13:
The `PySide6` and `pivy` are actually installed and working in the host env:

```
Python 3.11.10 (main, Sep  7 2024, 01:03:31) [GCC 13.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import PySide6 as p
>>> p.__version__
'6.8.0.2'
>>> import pivy
>>> pivy.__version__
'0.6.9'
>>> 
```

The compilation is ok. but FreeCAD can't locate the modules when ran.

mnesarco, 2024-12-13:
@looooo maybe you have a clue about this.

looooo, 2024-12-13:
No sry, I have no clue. Seems like everything is correctly setup.

mnesarco, 2024-12-14:
@looooo 
Yes it is something with the python packages path. I passed all paths manually (with a script) and it works, I don't know why FreeCAD does not detect them. I am inside a nix devShell, calling python from the shell can import any package, but FreeCAD can't because it only detects the path of the main python not the packages:

```
build/bin/FreeCAD \
-P/nix/store/virb8hapjj9v0gxdinxq4zp1qjba7ha0-audit-4.0/lib \
-P/nix/store/virb8hapjj9v0gxdinxq4zp1qjba7ha0-audit-4.0/lib/python3.12/site-packages \
-P/nix/store/90jj9wiqvhjgfmlxn96fqph54yaha9m4-gmsh-4.13.1/lib \
-P/nix/store/90jj9wiqvhjgfmlxn96fqph54yaha9m4-gmsh-4.13.1/lib/python3.12/site-packages \
-P/nix/store/879455nc7zqyf983xq0zz1vicgq93vy4-libcamera-0.3.2/lib \
-P/nix/store/879455nc7zqyf983xq0zz1vicgq93vy4-libcamera-0.3.2/lib/python3.12/site-packages \
-P/nix/store/rp6b82pm6b7x03ngbzffsl3vj8pvay5v-libselinux-3.7/lib \
-P/nix/store/rp6b82pm6b7x03ngbzffsl3vj8pvay5v-libselinux-3.7/lib/python3.12/site-packages \
-P/nix/store/s0xfjkpz3d9km7729bvb7x965kzrgkdh-lilv-0.24.24/lib \
-P/nix/store/s0xfjkpz3d9km7729bvb7x965kzrgkdh-lilv-0.24.24/lib/python3.12/site-packages \
-P/nix/store/37lw270b4ak30rvf77wrbzdai8j0saai-pyside6-6.8.0.2/lib \
-P/nix/store/37lw270b4ak30rvf77wrbzdai8j0saai-pyside6-6.8.0.2/lib/python3.12/site-packages \
-P/nix/store/xdpdf1njfzs6h87n2rbnhivm4xjrjdfk-python3-3.12.7-env/lib \
-P/nix/store/xdpdf1njfzs6h87n2rbnhivm4xjrjdfk-python3-3.12.7-env/lib/python3.12/site-packages \
-P/nix/store/nqdlpd1956gg4rzwnpz27z60ncy3bbl9-python3.12-autocommand-2.2.2/lib \
-P/nix/store/nqdlpd1956gg4rzwnpz27z60ncy3bbl9-python3.12-autocommand-2.2.2/lib/python3.12/site-packages \
-P/nix/store/sq70gv5jv52dkrm1m72ghpjv5ykr1yab-python3.12-contourpy-1.3.0/lib \
-P/nix/store/sq70gv5jv52dkrm1m72ghpjv5ykr1yab-python3.12-contourpy-1.3.0/lib/python3.12/site-packages \
-P/nix/store/w1cpg51q06km0mnl3vdz9s3wzs9a1icz-python3.12-cycler-0.12.1/lib \
-P/nix/store/w1cpg51q06km0mnl3vdz9s3wzs9a1icz-python3.12-cycler-0.12.1/lib/python3.12/site-packages \
-P/nix/store/chp5d36cxpg9ghd2vkkml9kpd16062xn-python3.12-distutils-75.1.1/lib \
-P/nix/store/chp5d36cxpg9ghd2vkkml9kpd16062xn-python3.12-distutils-75.1.1/lib/python3.12/site-packages \
-P/nix/store/511iv28iyxdw7prg6kj8896nkb6jgw25-python3.12-fonttools-4.54.1/lib \
-P/nix/store/511iv28iyxdw7prg6kj8896nkb6jgw25-python3.12-fonttools-4.54.1/lib/python3.12/site-packages \
-P/nix/store/whfw46n4jazawx175aynf99ww9jbjq7p-python3.12-inflect-7.4.0/lib \
-P/nix/store/whfw46n4jazawx175aynf99ww9jbjq7p-python3.12-inflect-7.4.0/lib/python3.12/site-packages \
-P/nix/store/llxxjh0cfyy5hc5jpvkyhx5z3765nn2p-python3.12-jaraco-classes-3.4.0/lib \
-P/nix/store/llxxjh0cfyy5hc5jpvkyhx5z3765nn2p-python3.12-jaraco-classes-3.4.0/lib/python3.12/site-packages \
-P/nix/store/4l1gvrzan659rj92sy69i4adav0194gb-python3.12-jaraco-collections-5.1.0/lib \
-P/nix/store/4l1gvrzan659rj92sy69i4adav0194gb-python3.12-jaraco-collections-5.1.0/lib/python3.12/site-packages \
-P/nix/store/g95khskb9yabrlqy7pfq97pkfzlfgy0m-python3.12-jaraco-context-6.0.1/lib \
-P/nix/store/g95khskb9yabrlqy7pfq97pkfzlfgy0m-python3.12-jaraco-context-6.0.1/lib/python3.12/site-packages \
-P/nix/store/fwwzch0yyxcya53q1qiiq5a0h3xbwgkb-python3.12-jaraco-functools-4.1.0/lib \
-P/nix/store/fwwzch0yyxcya53q1qiiq5a0h3xbwgkb-python3.12-jaraco-functools-4.1.0/lib/python3.12/site-packages \
-P/nix/store/vmd1j83klnibhcz3xk9nknaflmnf53a2-python3.12-jaraco-text-3.14.0/lib \
-P/nix/store/vmd1j83klnibhcz3xk9nknaflmnf53a2-python3.12-jaraco-text-3.14.0/lib/python3.12/site-packages \
-P/nix/store/bh163cg10x67d707cc3s2svv57yd2mpk-python3.12-kiwisolver-1.4.7/lib \
-P/nix/store/bh163cg10x67d707cc3s2svv57yd2mpk-python3.12-kiwisolver-1.4.7/lib/python3.12/site-packages \
-P/nix/store/xbn1rcbkkz641i21f21h0xsh6v6smsc2-python3.12-matplotlib-3.9.2/lib \
-P/nix/store/xbn1rcbkkz641i21f21h0xsh6v6smsc2-python3.12-matplotlib-3.9.2/lib/python3.12/site-packages \
-P/nix/store/z16hp32mnziffkci0wynbqbwdk1nf9cx-python3.12-more-itertools-10.5.0/lib \
-P/nix/store/z16hp32mnziffkci0wynbqbwdk1nf9cx-python3.12-more-itertools-10.5.0/lib/python3.12/site-packages \
-P/nix/store/f6n37s0wrqbiykxklxnm7bs82kap1f5p-python3.12-numpy-1.26.4/lib \
-P/nix/store/f6n37s0wrqbiykxklxnm7bs82kap1f5p-python3.12-numpy-1.26.4/lib/python3.12/site-packages \
-P/nix/store/qnfndbdg0k5s7zb47l5q9h7cszi56gpf-python3.12-packaging-24.1/lib \
-P/nix/store/qnfndbdg0k5s7zb47l5q9h7cszi56gpf-python3.12-packaging-24.1/lib/python3.12/site-packages \
-P/nix/store/gpa376dl0r8pwjjbfrj58xrwymq8nnli-python3.12-pillow-11.0.0/lib \
-P/nix/store/gpa376dl0r8pwjjbfrj58xrwymq8nnli-python3.12-pillow-11.0.0/lib/python3.12/site-packages \
-P/nix/store/dvkdvl5xkv5avj1b66ldwdjcwxm9ydw1-python3.12-pivy-0.6.9/lib \
-P/nix/store/dvkdvl5xkv5avj1b66ldwdjcwxm9ydw1-python3.12-pivy-0.6.9/lib/python3.12/site-packages \
-P/nix/store/72gj7lvii0cc6yc58zbsfrh8pg5n7nwq-python3.12-pybind11-2.13.6/lib \
-P/nix/store/72gj7lvii0cc6yc58zbsfrh8pg5n7nwq-python3.12-pybind11-2.13.6/lib/python3.12/site-packages \
-P/nix/store/rcsa1zaqrpndv37hz8m51ky8g93ddawl-python3.12-pyparsing-3.1.4/lib \
-P/nix/store/rcsa1zaqrpndv37hz8m51ky8g93ddawl-python3.12-pyparsing-3.1.4/lib/python3.12/site-packages \
-P/nix/store/iz14w235v0nmwc3vnzrlbrkfpaj1qqka-python3.12-python-dateutil-2.9.0.post0/lib \
-P/nix/store/iz14w235v0nmwc3vnzrlbrkfpaj1qqka-python3.12-python-dateutil-2.9.0.post0/lib/python3.12/site-packages \
-P/nix/store/h25lasfc88hing94ks429x393chr9jci-python3.12-setuptools-75.1.1/lib \
[... cut ...]
-P/nix/store/n2pgidpi7vpq2hbpylfpa4ck7avniipp-python3.12-typing-extensions-4.12.2/lib \
-P/nix/store/n2pgidpi7vpq2hbpylfpa4ck7avniipp-python3.12-typing-extensions-4.12.2/lib/python3.12/site-packages \
-P/nix/store/nxwbsz0q8m7hjga8q0f0ckmwcnhb9v8c-shiboken6-6.8.0.2/lib \
-P/nix/store/nxwbsz0q8m7hjga8q0f0ckmwcnhb9v8c-shiboken6-6.8.0.2/lib/python3.12/site-packages \
-P/nix/store/z7qd3dgvp4p7phismsgjaxnk79mbxxm0-speech-dispatcher-0.11.5/lib \
-P/nix/store/z7qd3dgvp4p7phismsgjaxnk79mbxxm0-speech-dispatcher-0.11.5/lib/python3.12/site-packages

```

mnesarco, 2024-12-14:
Hello, I nailed the nix flake to build it in linux. It works now on my laptop, I shared my flake here: https://github.com/mnesarco/freecad-nix-devshell

I am a nix noob, so maybe not the best approach but at least it compiles and runs.

```bash
$ git clone https://github.com/mnesarco/freecad-nix-devshell.git
$ cd freecad-nix-devshell
$ nix develop
$ clone
$ config
$ compile -j8
$ freecad
```

3x380V, 2024-12-27:
Since d01ce32a1f6d ("Update for LibPack3 (#10337)") `find_pip_package` sets output variables as `CACHE`:
https://github.com/FreeCAD/FreeCAD/blob/7083362acb8529017b58ea03bf7416da0192b25b/cMake/FreeCadMacros.cmake#L308-L316
So if a variable is already defined, this macro does not change it. This is true at least for `${PACKAGE}_FOUND` variable which is set to `FALSE` by previous call to `find_package`, so `PrintFinalReport` then reports PySide and Shiboken as not found.
@chennes, why are output variables tagged as `CACHE`?

mosfet80, 2025-03-03:
try this https://github.com/FreeCAD/FreeCAD/pull/19967

3x380V, 2025-03-10:
@iplayfast, care to re-test with #20107?

iplayfast, 2025-03-12:
I built it from scratch again, It compiled without errors but...
The issue persists.
 bin/FreeCAD
FreeCAD 1.1.0, Libs: 1.1.0devR40711 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

During initialization the error "No module named 'PySide6'" occurred in /home/chris/github/FreeCAD/build/Mod/Tux/InitGui.py
Please look into the log file for further information
During initialization the error "No module named 'PySide6'" occurred in /home/chris/github/FreeCAD/build/Mod/BIM/InitGui.py
Please look into the log file for further information
During initialization the error "No module named 'PySide6'" occurred in /home/chris/github/FreeCAD/build/Mod/Assembly/InitGui.py
Please look into the log file for further information
During initialization the error "No module named 'PySide6'" occurred in /home/chris/github/FreeCAD/build/Mod/AddonManager/InitGui.py
Please look into the log file for further information
During initialization the error "No module named 'PySide6'" occurred in /home/chris/github/FreeCAD/build/Mod/Measure/InitGui.py
Please look into the log file for further information
During initialization the error "No module named 'PySide6'" occurred in /home/chris/github/FreeCAD/build/Mod/Help/InitGui.py
Please look into the log file for further information
During initialization the error "No module named 'PySide6'" occurred in /home/chris/github/FreeCAD/build/Mod/Draft/InitGui.py
Please look into the log file for further information
During initialization the error "No module named 'PySide6'" occurred in /home/chris/.local/share/FreeCAD/Mod/btl/./InitGui.py
Please look into the log file for further information
Wizard shaft module cannot be loaded
No module named 'PySide6'
Wizard shaft module cannot be loaded
No module named 'PySide6'
Wizard shaft module cannot be loaded
No module named 'PySide6'
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
[... 92 lines cut ...]
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap
QPixmap::scaled: Pixmap is a null pixmap

nerozero, 2025-06-08:
Just updated and compiled from git. Facing similar issue while trying to run fresh compiled version... 
Any solution ?

nerozero, 2025-06-08:
@chennes sure

```sh
cmake ../freecad-source
-- Using ccache found at: /usr/bin/ccache
-- Compiler: GNU, version: 13.3.0
-- prefix: /usr/local
-- bindir: bin
-- datadir: share
-- docdir: share/doc/FreeCAD
-- includedir: include
-- libdir: lib
-- cmake: 3.28.3
-- Compiling with Qt 6
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
=====================================================
Doxygen not found, will not build documentation.
=====================================================

-- find_package() was used to locate fmt version 9.1.0
-- Found Python3: /usr/bin/python3 (found version "3.12.3") found components: Interpreter Development Development.Module Development.Embed
-- Found pybind11: /usr/include (found version "2.11.1")
-- PyCXX found:
--   Headers:  /home/nerozero/Programs/compile/freecad/freecad-source/src/3rdParty/PyCXX
--   Sources:  /home/nerozero/Programs/compile/freecad/freecad-source/src/3rdParty/PyCXX/CXX
--   Version:  7.1.9
-- Found OpenCASCADE version: 7.6.3
--   OpenCASCADE include directory: /usr/include/opencascade
--   OpenCASCADE shared libraries directory: /usr/lib/x86_64-linux-gnu
-- VTK components: CommonCore;CommonDataModel;FiltersVerdict;IOXML;FiltersCore;FiltersGeneral;IOLegacy;FiltersExtraction;FiltersSources;FiltersGeometry;WrappingPythonCore;IOMPIParallel;ParallelMPI;hdf5;RenderingCore;InteractionStyle;RenderingFreeType;RenderingOpenGL2
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so  found components: OpenGL GLX
-- Found Python3: /usr/bin/python3 (found suitable version "3.12.3", minimum required is "3.12") found components: Interpreter Development.Module Development.Embed
-- Found HDF5: hdf5;/usr/lib/x86_64-linux-gnu/hdf5/serial/libhdf5.so;/usr/lib/x86_64-linux-gnu/libcrypto.so;/usr/lib/x86_64-linux-gnu/libcurl.so;/usr/lib/x86_64-linux-gnu/libpthread.a;/usr/lib/x86_64-linux-gnu/libsz.so;/usr/lib/x86_64-linux-gnu/libz.so;/usr/lib/x86_64-linux-gnu/libdl.a;/usr/lib/x86_64-linux-gnu/libm.so (found version "1.10.10") found components: C HL
-- Found Eigen3: /usr/include/eigen3 (found version "3.4.0")
-- VTK python wrapper: available
-- Check for medfile (libmed and libmedc) ...
-- We guess that libmed was built using hdf5-serial version
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so
-- Set up to compile with Qt 6.4.2
-- Checking Pivy version by importing it in a Python program...
-- Found Pivy 0.6.9
-- Checking Pivy Coin3D version by importing it in a Python program...
-- Found Pivy Coin3D SIM Coin 4.0.2
WARNING: Skipping /usr/lib/python3.12/dist-packages/pybind11-2.11.1.dist-info due to invalid metadata entry 'name'
WARNING: Package(s) not found: Shiboken6
WARNING: Skipping /usr/lib/python3.12/dist-packages/pybind11-2.11.1.dist-info due to invalid metadata entry 'name'
WARNING: Package(s) not found: shiboken6_generator
WARNING: Skipping /usr/lib/python3.12/dist-packages/pybind11-2.11.1.dist-info due to invalid metadata entry 'name'
WARNING: Package(s) not found: Shiboken6
WARNING: Skipping /usr/lib/python3.12/dist-packages/pybind11-2.11.1.dist-info due to invalid metadata entry 'name'
WARNING: Package(s) not found: PySide6
-- Found PYSIDE6 tools: /usr/lib/qt6/libexec/uic, /usr/lib/qt6/libexec/rcc
CMake Warning at cMake/FreeCAD_Helpers/SetupShibokenAndPyside.cmake:128 (message):
  Shiboken6 include files not found, FREECAD_USE_SHIBOKEN automatically set
  to OFF
Call Stack (most recent call first):
  CMakeLists.txt:111 (SetupShibokenAndPyside)

Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'shiboken6'
CMake Warning at cMake/FreeCAD_Helpers/SetupShibokenAndPyside.cmake:140 (message):
  ==================================

  Shiboken6 Python module not found.

  ==================================

Call Stack (most recent call first):
  CMakeLists.txt:111 (SetupShibokenAndPyside)
[... 157 lines cut ...]
-- QtOpenGL:                    6.4.2
-- QtPrintSupport:              6.4.2
-- QtSvg:                       6.4.2
-- QtTest:                      6.4.2
-- QtUiTools:                   6.4.2
-- QtWidgets:                   6.4.2
-- QtXml:                       6.4.2
-- Rift:                        not enabled (BUILD_VR)
-- Shiboken:                    not found
-- SMESH:                       7.7.1.0 build internal
-- SPNAV:                       [/usr/lib/libspnav.so] [/usr/include]
-- SWIG:                        4.2.0
-- Vtk:                         9.1.0
-- XercesC:                     3.2.4 [/usr/lib/x86_64-linux-gnu/libxerces-c.so] [/usr/include]
-- yaml-cpp:                    0.8.0
-- ZLIB:                        1.3

=================================================
Now run 'cmake --build /home/nerozero/Programs/compile/freecad/freecad-build' to build FreeCAD
=================================================

-- Configuring done (3.8s)
-- Generating done (1.2s)
-- Build files have been written to: /home/nerozero/Programs/compile/freecad/freecad-build//
```

closed 2025-09-21 by github-actions[bot]