FreeCAD/FreeCAD #18105: FC1.0.0 / PartDesign : thread modeled prevents a Pad

Almisuifre, 2024-11-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When attempting to create a hole using the fillet feature, an error occurs if a new sketch is added to add material. The error message displayed is: "Removing Spliter Filed".

This issue impacts the workflow by preventing users from modifying or adding material to designs after using the fillet operation, disrupting design flexibility and potentially increasing the time needed to troubleshoot or work around the issue.

The problem has been reproduced on both Windows 11 and Linux Mint.

Create a feature with a fillet.
Add a new sketch to introduce additional material to the model.
Observe the error message: "Removing Spliter Filed".

### Full version info

```shell
The issue was reproduced using the following configuration:
On Windows 11:
OS: Windows 11 build 22631  
Word size of FreeCAD: 64-bit  
Version: 0.21.2.33771 (Git)  
Build type: Release  
Branch: releases/FreeCAD-0-21  
Hash: b9bfa5c5507506e4515816414cd27f4851d00489  
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3  
Locale: French/France (fr_FR)  

On Linux Mint:

Similar behavior was observed under Linux Mint, using a comparable FreeCAD version.

Testing the issue in safe mode on both systems did not resolve the problem.
```

### Subproject(s) affected?

PartDesign

### Anything else?

Additional Context

A discussion about this issue is available on the FreeCAD forum, including screenshots illustrating the problem:
[Forum thread link](https://forum.freecad.org/viewtopic.php?t=92329).

For further diagnosis, the related .FCStd file and additional logs can be provided upon request.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-11-24:
I dont see any fillet operation in your example.
Issue present if **Model thread** switched on.

FlachyJoe, 2024-11-24:
**Removing Spliter Failed** comes with a refinement error. Try to disable the Refine property of this feature and/or of the previous ones.

tarman3, 2024-11-24:
Disable Refine in all operation.
Issue is still present.

Almisuifre, 2024-11-24:
I don’t see a refine function. I use FreeCAD with its basic functions.

tarman3, 2024-11-24:
[image]

Almisuifre, 2024-11-24:
Ok, that’s good news to pass on.

But, that said, it would be fine if it is, is disabled by default (although I do not recommend this).

Or is it planned to be fixed in a future version?

What did the programmers originally plan in V1.0.0?

Almisuifre, 2024-11-24:
I just tried this method: Refine = False.

Result, it does not work at all!
[Turbine.zip](https://github.com/user-attachments/files/17893737/Turbine.zip)

FlachyJoe, 2024-11-24:
@Almisuifre disable this Part/Part Design settings and retry
[image]

Almisuifre, 2024-11-25:
These are the options that work for me.
[image]

drcowsley, 2024-12-15:
I get the same message - this time when adding a pocket (Pocket 004
[image]
[RemovingSplitterFails.zip](https://github.com/user-attachments/files/18142048/RemovingSplitterFails.zip)