FreeCAD/FreeCAD #18106: Mirror tool on Part workbench

DrAvriLev, 2024-11-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Mirror command malfunction

### Full version info

```shell
This is my config:
OS: Windows 11 build 26120
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Hebrew/Israel (he_IL)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Fusion
Installed mods: 
  * BIM021 2021.12.0
  * Curves 0.6.51
  * Freecad-Built-in-themes-beta 1.2.2
  * FreeCAD-themes 2024.7.24
  * lattice2 1.0.0
  * Manipulator 1.5.7
  * ProDarkThemePreferencePack 1.0.0
```

### Subproject(s) affected?

None

### Anything else?

I'm learning FreeCad and designed this:
[image]

Now, when trying to mirror it I get this:
[image]

Only using the transform command I end up with the correct body:
[image]

I think it should do it automatically, without me using the transform command.

Any suggestions?

Avri

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-11-24:
Looks like you set not correct mirror plane.
Please attache file.

DrAvriLev, 2024-11-24:
I cannot add a file of .FCStd
How can I add it?

DrAvriLev, 2024-11-24:
[Part_03.zip](https://github.com/user-attachments/files/17893067/Part_03.zip)

tarman3, 2024-11-24:
[Screencast_1.webm](https://github.com/user-attachments/assets/f184bb7c-9280-486f-bdb4-5991892353d6)

tarman3, 2024-11-24:
If you had immediately created half the model relative to the base plane, then creating the datum plane could have been avoided.

[image]

tarman3, 2024-11-24:
Combine **Part** and **PartDesign** should always be a last resort.
Never leave the PartDesign workbench unless you know exactly what you are doing.

DrAvriLev, 2024-11-24:
Thank you tarman3, I used a datum plane as you suggested. It worked just fine.