FreeCAD/FreeCAD #18107: IFC export problem when objects are placed inside of a group

semhustej, 2024-11-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I encounter problem with Export to IFC when objects are placed inside of a group. Objects are not cut properly (no openings, roof does not cut walls).

Steps to reproduce:

- Open file 
[025_IFC_export-group.FCStd.zip](https://github.com/user-attachments/files/17893009/025_IFC_export-group.FCStd.zip)

- Select all objects in the 3D view using box select (walls, windows and roof are placed in the group 'Construction')
[image]

- File -> Export....    and select IFC
- Open the IFC in IFC viewer (I tried BIMvision and BonsaiBIM) -> Walls are **not** cut by the roof, Walls **do not have** window and door openings:
[image]

-----------------------------

When BIM objects (walls, etc.) are not placed in the 'Construction' group, the cuts work OK.
Steps to reproduce:

- Open the sample file
- Select all objects in 'Construction' group and move them out of the group to the Level01 object
- Select all objects in the 3D view using box select
[image]

- File -> Export....    and select IFC
- Open the IFC in IFC viewer (I tried BIMvision and BonsaiBIM) -> Walls are cut by the roof, Walls have window and door openings:
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

BIM

### Anything else?

In the second workflow case, there is a separate issue, where the Roof is not exported. Should this be reported as a separate issue or can it stay here?

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

paullee0, 2024-11-24:
Roof is still missing as you reported in FC forum, right?

And, apparently Construction group  is 'for construction' thus ignored by the code - just guessing, not checked the code :)

semhustej, 2024-11-24:
> Roof is still missing as you reported in FC forum, right?
> 
Yes, not sure if I should report this as a separate issue.

> And, apparently Construction group is 'for construction' thus ignored by the code - just guessing, not checked the code :)
> 
> (EDIT Forum Discussion - https://forum.freecad.org/viewtopic.php?p=793681#p793681)

I tried renaming the group to 'Group'. The result is the same. The name of the group has no influence on the output. Sorry for the confusing name, this did not occur to me previously.

yorikvanhavre, 2024-11-25:
This should be fixed by https://github.com/FreeCAD/FreeCAD/pull/17004
It has now been merged, if someone could test again?

semhustej, 2024-12-08:
The export with latest weekly produces the same result. It seems to be not fixed on my side.
[image]

The same result is produced with regards to walls not being cut when switching to strict IFC mode.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39503 (Git) Conda
Build type: Release
Branch: main
Hash: da45328f6f9b35cbbdc7167d7faadb6a07634bdb
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

```

luzpaz, 2024-12-11:
@yorikvanhavre JFYI :point_up_2:

yorikvanhavre, 2024-12-13:
We need to redo the export system there... This should be fixed together with fixing #17862

yorikvanhavre, 2024-12-13:
Duplicate of #17862

semhustej, 2025-03-22:
Just a note to test after the fix has been applied:

I found out, that classification of objects is not exported ti IFC if those objects are inside of a group (such as the OP walls).

furgo16, 2025-05-02:
I'm at a point where I'm attempting to export all my models to IFC, and I can confirm this bug is still present.

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.41488 (Git) Snap 1432
Build date: 2025/05/01 22:16:31
Build type: Release
Branch: main
Hash: b3a3b13603e7c48c6349080a893471ae9c105bc1
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 9.1.0, boost 1_74, Eigen3 3.4.0, PySide 5.15.11
shiboken 5.15.11, SMESH 7.7.1.0, xerces-c 3.2.3, IfcOpenShell 0.8.0, OCC 7.7.1
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/159.34/1
Installed mods: 
  * Telemetry 1.0.1
  * Assembly4 0.50.17
  * Launcher
```

furgo16, 2025-05-17:
@semhustej while testing https://github.com/FreeCAD/FreeCAD/pull/21409 and using the attached garage test file there, I could no longer reproduce this issue. Could you retest with the change from that PR? Thanks!

furgo16, 2025-05-17:
Nevermind, false positive. I had tested it with the .dae export, which looked good, but the .ifc export still exhibits the same behaviour described on this issue:

[image]

Questions: how do FreeCAD groups translate into IFC entities? If a model is expected to be exported to IFC, should it use groups at all?

furgo16, 2025-05-17:
> Questions: how do FreeCAD groups translate into IFC entities? If a model is expected to be exported to IFC, should it use groups at all?

I cannot answer the question, but I've noticed:
- If groups are inside a level in the FreeCAD file, they are NOT exported to the IFC file
- If groups are outside a level in the FreeCAD file, they are exported as `IfcElementAssembly`

[image]

[image]

furgo16, 2025-05-17:
Regarding substraction volumes, this piece of code might be relevant:

https://github.com/FreeCAD/FreeCAD/blob/1ca108f8d503b2b68455844bfdfabb62fa1461ed/src/Mod/BIM/importers/exportIFC.py#L644-L676

semhustej, 2025-05-17:
@furgo16 you were quicker :). I posted in this thread just now: https://forum.freecad.org/viewtopic.php?t=97111

semhustej, 2025-05-17:
> * If groups are inside a level in the FreeCAD file, they are NOT exported to the IFC file

I can not confirm this in my test. The group is exported even though it is inside of `Level` 

FreeCAD:

[image]

Open IFC Viewer:

[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41694 (Git)
Build date: 2025/05/13 12:46:54
Build type: Release
Branch: main
Hash: 5e624a4775c993a3d940f37876840c65ac0cc60c
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25
Installed mods: 
  * Reporting
  * Telemetry 1.0.1

```

furgo16, 2025-09-06:
Still relevant, although there is a working workaround.