FreeCAD/FreeCAD #18108: Additive Helix with negative Cone Angle breaks with height over one revolution

schulzh, 2024-11-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I am trying to edit this model in FC 1.0: https://www.printables.com/model/826091

Here is what the file looks like in 0.21.2: 
[image]

When opening the file in 1.0 the threads are reversed, they go down and expand outward. Also the settings of the helix:
[image]

Playing around with the settings, the helix is correct up to (not including) a height of 6.0mm, which is exactly one revolution:
[image]

### Full version info

```shell
OS: KDE Flatpak runtime (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * fasteners 0.5.31

Also tried this on Windows, same result
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-24:
@schulzh could it be the same as https://github.com/FreeCAD/FreeCAD/issues/18072
@FlachyJoe FYI

schulzh, 2024-11-24:
I don't think so, I don't get any error message related to the helix (it just computes wrong) and the linked issue states that cone angle 0 does not work, which is not the case in this model.

tarman3, 2024-11-24:
I confirm.
The behavior is strange. At a negative angle, the direction of the spiral changes.
I am not sure how it was before 1.0

[image]

<details>
  <summary>FreeCAD 1.0.0.39109 (Git), Arch Linux (KDE/plasma/wayland)</summary>

```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.31; sheetmetal 0.5.7; Silk 0.1.5; Curves 0.6.51
```
</details>

<details>
  <summary>FreeCAD 1.1.0dev.39240 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39240 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: f9ba94c70b139ece6ba5ffba47632d91b5c9c645
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.31; sheetmetal 0.5.7; Silk 0.1.5; Curves 0.6.51
```
</details>

[image]

FlachyJoe, 2024-11-24:
I think it is a previously fixed bug (before the TNP mitigation), the fix was certainly lost.
Can someone search on closed issues?

maxwxyz, 2024-11-24:
@FlachyJoe yes: https://github.com/FreeCAD/FreeCAD/issues/10284

schulzh, 2024-11-24:
I've noticed another strange thing about this bug: In case there is more than one revolution, the helix actually extrudes in the correct direction for the remainder until any full revolution:
[image]
[image]
[image]

tarman3, 2024-11-24:
https://github.com/user-attachments/assets/52629f67-c918-4344-ae57-7ea3792f038d

luzpaz, 2024-12-10:
PR is ready to merge

tarman3, 2024-12-17:
In last dev negative cone angle still mirroring helix if cross-section is circle.
With rectangle no issue. 

https://github.com/user-attachments/assets/32f6235c-649b-4e7c-b5e9-526d90780368

luzpaz, 2024-12-18:
@FlachyJoe :point_up_2:

FlachyJoe, 2024-12-18:
@luzpaz @tarman3 Grlmbrgrlm ! 😤

FlachyJoe, 2024-12-21:
The problem occurs when the profile center is on the sketch X axis. It can be reproduced with a centered rectangle too. I still investigate.

FlachyJoe, 2025-01-07:
If the profile center is below the 179.6725° construction line then the helix fails.
[image]

The computed path is good so it seems an OCC bug.

@AIRCAP may it be solved with another tolerance fix ?

leoheck, 2025-04-18:
Now I understand the issue better. 

Maybe it would be good that the user could have an warning or an info message or have the tooltip of the cone angle stating that negative values can make the helix/thread/feature self interact and then produce unexpected/broken results?

binomnom, 2025-05-10:
> I confirm. The behavior is strange. At a negative angle, the direction of the spiral changes. I am not sure how it was before 1.0

I found this issue because I am encountering the same problem (Version: 1.0.0.39109 (Git), Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20)

> If the profile center is below the 179.6725° construction line then the helix fails.

I could not reproduce this behaviour with my more complex profile. No matter which Quadrant I put it in, it always does this.

[PC4-01_Nut_BSPT_Rc1-8.zip](https://github.com/user-attachments/files/20140748/PC4-01_Nut_BSPT_Rc1-8.zip)

binomnom, 2025-05-10:
I work around this bug for now by just starting the helix from the other end.

I have found another weird behavior with the taper angle in PartDesign Helix that I'd like to show here, so you can tell me whether this is the same issue or if I should open another issue for it:
[PC4-01_Nut_BSPT_Rc1-8_V2.zip](https://github.com/user-attachments/files/20140800/PC4-01_Nut_BSPT_Rc1-8_V2.zip)

If and only if there is a value other than 0 in the taper field, the helix fails as soon as it reaches an edge in the tip's geometry:

Height set so that edge of the object is not intersected:
[image]

Increased height by 0.1mm, helix reaches edge and fails:
[image]

But when angle is reset to 0, helix works without issue:
[image]

What do you think of this?

FlachyJoe, 2025-05-11:
@binomnom 

1. disable the auto-refine preferences when you work with helix: #17846
2. switch off all the Refine properties of your model
3. **PC4-01_Nut_BSPT_Rc1-8_v3** is now working when you set **Helix001.SegmentLength=1**

I've not tested your other files but I'm confident Refine is the major issue.

binomnom, 2025-05-11:
@FlachyJoe thank you for taking the time to have a look! So I am not doing anything fundamentally wrong, but the issue is well known. Very grateful for your help!

pieterhijma, 2025-08-31:
I check with the latest main:

```
OS: Arch Linux (tty)
Architecture: x86_64
Version: 1.1.0dev.43162 (Git)
Build date: 2025/08/31 03:10:14
Build type: Unknown
Branch: main
Hash: 9d806694ffe4245e2d6ad11abda406a9b076872f
Python 3.13.5, Qt 6.9.1, Coin 4.0.3, Vtk 9.5.0, boost 1_88, Eigen3 3.4.0, PySide 6.9.1
shiboken 6.9.1, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.9.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1.33333
Installed mods: 
  * OpenTheme 2024.5.16
  * Ondsel-Lens 2024.10.21.01
  * fasteners 0.5.34
  * WebTools 1.0.0
  * osh-autodoc-workbench 0.2.3
  * DynamicData 2.70.0
  * Curves 0.6.61
  * freecad.gears 1.3.0
  * sheetmetal 0.7.24
```

The fusion seems to give problems in rendering, but if I check the geometry, then all is fine. Before the fusion:

[image]

After the fusion:

[image]

The same is true for the whole screw:

[image]

However, if I export the screw as a brep and then import it as a brep, all looks good:

[image]

There are no errors when I do Part -> Check geometry.

What to do with this issue?

pieterhijma, 2025-09-01:
> when we are going to have the occt upgraded?

That is a challenging question. Linux distributions can ship any OCCT they deem fit. We are planning to target the latest stable from OpenCascade for the full builds that we do ourselves but that hasn't happened yet.