FreeCAD/FreeCAD #18109: Start : Thumbnail is not displayed if name of file contains dot

VincidaB, 2024-11-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Thumbnail is not displayed if name of file contains `.` in it's name.
[image]

I have tracked this down to this line : 
https://github.com/FreeCAD/FreeCAD/blob/45134cef165ecc11aca69f81673e7b69421cebf1/src/Mod/Start/App/DisplayedFilesModel.cpp#L282

I can make a PR changing this to `suffix()`. Is there a reason for using `completeSuffix()` ? 

### Full version info

```shell
OS: Ubuntu 24.10 (xcb)
Architecture: x86_64
Version: 1.1.0dev.39258 +1 (Git)
Build type: Debug
Branch: pr/16414
Hash: 479823a9efab9f8b77283fd1b48496cc90703d3f
Python 3.12.7, Qt 5.15.15, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-11-24:
@wwmayer can you weigh-in ?

pburger24, 2024-11-25:
- I'm not sure if this should be written as a separate issue but a similar issue occurs with files that dont end in .FCStd

[image]

Both of these files open the same and both work, but one is not showing the thumbnail

wwmayer, 2024-11-26:
> Both of these files open the same and both work, but one is not showing the thumbnail

The thumbnail of the .FCStd is part of the file itself and can be easily extracted. For a STEP file one would have to open FreeCAD, load the file, make a screenshot and unload the file. This would be a very expensive and troublesome operation and thus not doable.

commit 3070832e (2024-11-26): Start: Show thumbnail if filename contains a dot

Fixes #18109
closed 2024-11-26 by chennes