FreeCAD/FreeCAD #18111: Backward compatibility with openGL 1.1 ?

ClubACNews, 2024-11-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi,
I was hoping to use Freecad and Kicad with Windows 11 ARM on a Rapberry-pi 4B.
Kicad works well even 3D representation of PCB boards.
Freecad 1.00 does not work because it uses at least OpenGL 2 while the graphics chip only has an OpenGL 1.1 driver for Windows 11 under ARM. 
So, you can't go further than the home screen.
Please consider this future possibility.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * fasteners 0.5.29
  * OpenTheme 2024.9.1
  * ProDarkThemePreferencePack 1.0.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-24:
@adrianinsaval is this feasible or should this be closed in favor of https://github.com/FreeCAD/FreeCAD/issues/17181

hyarion, 2024-11-24:
OpenGL 1.1 was released 27 years ago. A lot has happened since then and there are features that freecad uses that requires a more modern opengl version.
I don't see it likely that freecad will move backwards to support a legacy opengl version, if anything I think freecad should move forward and use more modern frameworks.

@maxwxyz this is a separate issue from 17181, there are windows 11 arm64 devices that has higher opengl versions that should be compatible with freecad.