FreeCAD/FreeCAD #18112: Ifc merge port to freecad

Isaina, 2024-11-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Can you port Perl script to freecad ifc  project for ifcmerge 

### Full version info

```shell
So we can use GitHub for ifc 

https://www.youtube.com/watch?v=oRCtyrm2REs
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-11-25:
Having a merge tool is of course a good idea, but not really sure how this particular one would be implemented in FreeCAD. This needs more planning IMHO

brunopostle, 2024-12-01:
Just noting that that [ifcmerge](https://github.com/brunopostle/ifcmerge/) has to be a standalone executable due to the way git works, so porting to python isn't so critical. Also, merging is nice, but just being able to commit, tag, push, etc.. is really useful. I'm happy to help of course, there are lots of gnarly git and GitPython quirks that are handled in the Bonsai implementation.

Isaina, 2024-12-01:
So in Linux and macos perl is native almost can be just written plug-in for BIM interface which use IFc native porting code from BlenderBIM (Bonsai) basically for making easier 2d documentation and BonsaiBIm for visualization

yorikvanhavre, 2024-12-03:
We had/have a Git tool in FreeCAD already, in the WebTools addon: https://wiki.freecad.org/WebTools_Git

There might be some work needed there, if someone is interested in resurrecting it. I have the impression nobody used it for some time...

Isaina, 2026-11-13:
@tritao hi can this be added also in bim with 2D edit mode because blender lack this type of editor but has greater visualization project so we can work on one ifc file thru git