FreeCAD/FreeCAD #18113: Ability to easily convert a chamfer to a fillet

longrackslabs, 2024-11-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Often i'll create a fillet (or chamfer) in Part Design, then decide i would rather have a chamfer (or fillet).  How hard would it be to right click and covert from one to the other?  Otherwise I have to delete the feature and start over.

### Full version info

```shell
1.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-25:
You could argue that for each feature: pocket/pad, revolve/groove.
@FreeCAD/cad-advisory-group what is your take on that?

longrackslabs, 2024-11-25:
> In terms of engineering, chamfer situations are different from fillet
> situations. It is rare when both are valid. I don't think it is useful to
> provide that switch considering that the workaround is very easy. It
> doesn't exist in packages I know (Cathia, Solidworks, Alibre).

What's the easy workaround  (and not "delete one and build the other"?

pierreporte, 2024-11-25:
The deletion is the workaround.

longrackslabs, 2024-11-25:
That's not really a work around is it.

Anyway, more of a usability / UI suggestion.

closed 2025-09-20 by longrackslabs