FreeCAD/FreeCAD #18114: "Can't find Origin" Message when inserting Assembly containing Sub-Assemblies.

MansourMohamedAli, 2024-11-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I searched for this issue but did not find a post about it. Sorry if it already exists.

I wanted to create an assembly that uses another assembly multiple times. So, I did the following:

1. [TSLOT-1515S.FCStd] Created the parts.
2. [SubAssembly.FCStd] Created a new assembly and inserted those parts. 
3. [Multi-Subassembly.FCStd] Created a new assembly and inserted the assembly created in step 2 twice. 
4. [FullAssembly.FCStd] Finally, created another assembly and inserted the Multi-Subassembly into this Full Assembly.

The Multi-Subassembly was indeed inserted, however, Error messages appeared on screen and in Report View:

21:25:00  Can't find Origin for "FullAssembly#SubAssembly"
21:25:00  Can't find Origin for "FullAssembly#SubAssembly"
21:25:00  Can't find Origin for "FullAssembly#SubAssembly"
21:25:00  Can't find Origin for "FullAssembly#SubAssembly"
21:25:00  Can't find Origin for "FullAssembly#SubAssembly"
21:25:00  Can't find Origin for "FullAssembly#SubAssembly"
21:25:00  Can't find Origin for "FullAssembly#SubAssembly"
21:25:00  Can't find Origin for "FullAssembly#SubAssembly001"
21:25:00  Can't find Origin for "FullAssembly#SubAssembly001"
21:25:00  Can't find Origin for "FullAssembly#SubAssembly001"
21:25:00  Can't find Origin for "FullAssembly#SubAssembly001"
21:25:00  Can't find Origin for "FullAssembly#SubAssembly001"
21:25:00  Can't find Origin for "FullAssembly#SubAssembly001"
21:25:00  Can't find Origin for "FullAssembly#SubAssembly001"

Now I did notice the Origin would get renamed when inserting a previously made assembly (Origin would be renamed Origin001). I have worked on RC2 and I recall I was able to insert assemblies made of other assemblies with no issues. In that case, the Origin was not renamed to Origin001, Origin002, etc... but would maintain it original name of Origin. 

I attached screenshots and the FreeCAD files I used. To see the error, Open file FullAssembly.FCStd and insert "Multi-Subassembly.FCStd" into its assembly.

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * A2plus 0.4.65
```

### Subproject(s) affected?

Assembly

### Anything else?

[SubAssemblyIssue.zip](https://github.com/user-attachments/files/17896450/SubAssemblyIssue.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

lmurta, 2024-11-25:
I believe it is the same bug at https://forum.freecad.org/viewtopic.php?p=559682#p559682
I have the same problem and it can be solved with this workaround:

```
sel = Gui.Selection.getSelection()
origin = App.ActiveDocument.addObject('App::Origin', 'Origin')
App.ActiveDocument.getObject(sel[0].Name).Origin = origin #object Part selected in ComboView
App.activeDocument().recompute(None,True,True)
```

philjynx, 2024-12-06:
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * A2plus 0.4.68
  * ExplodedAssembly
  * fcgear
  * FreeCAD-themes 2024.7.24
  * freecad.gears 1.3.0
  * Help 1.0.3
  * HiResMonitor
  * kicadStepUpMod 11.4.3
  * Manipulator 1.5.7

Same here:
Pictured sub assembly (complete with origin) 

[image]

and 'Insert'ed component stripped of origin.

[image]

philjynx, 2024-12-06:
Some supplimentary info.
When choosing sub assemblies to insert I've noticed that the ones that are going to fail have their own subassemblies listed in the pane. The ones that do not fail do not have their subassemblies listed. All of the items I have scribbled on contain a minimum of two instances of the same BearingBlockAssembly (with only one source file for that assembly).

CarriageAssembly and ShortSlideAssy both fail with 'Can't find Origin' on the red circled subassemblies when inserted.
The three other assemblies (with the green pointer) insert with no complaint.

spiridonios, 2024-12-09:
I think a deal with the same problem. when i open the project i work with, i see many subassemblies having turn red with errors indicating an origin feature missing. 

Maybe it's worth noting that the problem appears high in the tree. I have a main assembly, called "assembly". Below it a have a "fusion" with a house's room designed, (all good there) and two more assemblies called "Assembly001" and "Assembly005", that represent two parts of a furniture. In them a have several more assemblies. When i open the files "Assembly001" and "Assembly005" there is no problem with errors, but when i open the parent "assembly" i see the errors on the drawers of the two furniture subassemblies ("Assembly001" and "Assembly005). I attach my working file here.
[krevati_peristera_03.tar.gz](https://github.com/user-attachments/files/18056819/krevati_peristera_03.tar.gz)

Other than the bugs (i guess the problem i also refer here is a bug and not something i dig wrong..) the new assembly workbench is a charm! Really easy and fun to work with! So hopefully as times goes on these bugs will be fixxed. thank you for all your hard effort freecad developers!

Now as i understand from this post and the linked freecad forum post, a script can be used in order to overcome this issue. I understand that i should click on the error part on the tree view and then run a macro. I try to understand how to do this, but i have zero experience with running macros and even after reading a little bit about them i am a bit confused of how exactly i should proceed with this. Could maybe someone drop a few hints of how i should proceed with this?

lmurta, 2024-12-09:
To apply the script/workaround you must have the Python Console opened: Menu View>Panels>Python Console:
[image]
Then, select the assembly with the problem/red mark, paste the script in the python console and click Enter.
[image]
Repeat for each assembly with the problem.

philjynx, 2024-12-09:
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * FreeCAD-themes 2024.7.24
  * Help 1.0.3
  * HiResMonitor
  * kicadStepUpMod 11.4.4
  * Manipulator 1.5.7

`sel = Gui.Selection.getSelection()
origin = App.ActiveDocument.addObject('App::Origin', 'Origin')
App.ActiveDocument.getObject(sel[0].Name).Origin = origin #object Part selected in ComboView
App.activeDocument().recompute(None,True,True)`

**Confirmed** Using the above macro does, indeed, insert the missing Origin.
_**Speculation**_ The missing origin might be the cause of the access violations, I've just been playing with Assemly WB for over an hour, running the macro when necessary to insert missing origins, making changes and additions with no crashes.

spiridonios, 2024-12-10:
thanks Imurta! following your advice, it seems that i have a file with no errors now! I saved with a new name and reopened the file and no errors appear!

philjynx, 2024-12-15:
Why closed? Confirmed issue.

RolleLight, 2025-01-07:
> I believe it is the same bug at https://forum.freecad.org/viewtopic.php?p=559682#p559682 I have the same problem and it can be solved with this workaround:
>
> ```
> sel = Gui.Selection.getSelection() [...]

Had the same problem, your code fixed that, thank you ! :-)

vpmier, 2025-01-12:
Hi,

The python script below fixes the error message for me too. Thanks !

I am using the assembly workbench for an assembly (we can call it level 1) that contains sub-assemblies (level 2) and sub-sub-assemblies (level 3).
The origin disappear only for the level 3 assemblies in the tree view of the level 1 assembly.
The origins of the level 3 assemblies are still present when we open the level 2 assemblies.
So:
- a difference of 1 level -> OK
- a difference of 2 levels -> disappearing of the level "N+1" assembly origin. I wonder why.

It might also be interesting to test more level of depth and see if every levels after the 3rd are affected.

samatthias, 2025-01-21:
Hello, I had the same problem with my project. The described method worked for me as well. See Freecad Forum entry: https://forum.freecad.org/viewtopic.php?t=94084

toenails6, 2025-05-20:
Also had this issue, the mentioned solution works. 
Hope this can be fixed in the next update though.

PR #22415 (2025-07-17): Assembly: Fixes bug for origins in Sub-`AssemblyLink`
Fixes #18114, #20270

When creating an `AssemblyLink` object, simply links the origin, as it did not before.