FreeCAD/FreeCAD #18115: Sketcher - B spline not constrained on circle, nor tangent.

NathanLKing, 2024-11-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I create a circle and try to add tangent constraint, it moves and applies. 
Yet, zoom in and it shows that it isn't. Try to pad and it throws the "Pad: Wire is not closed"
Which is shown in Sketcher. 

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Australia (en_AU)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Assembly4 0.50.15
  * fasteners 0.5.27
  * QuickMeasure 2022.10.28
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-11-25:
Please attach file

NathanLKing, 2024-11-25:
[image]

NathanLKing, 2024-11-25:
Another issue, this system won't allow attaching a freecad file, what the?

tarman3, 2024-11-25:
Pack FCStd file to zip

NathanLKing, 2024-11-25:
> Pack FCStd file to zip

Yes, crazy that I have to do that, especially because 7zip files aren't allowed either, 
[Heart.zip](https://github.com/user-attachments/files/17897673/Heart.zip)
but here it is.

tarman3, 2024-11-25:
The problem not with tangent. Your wire in the sketch is really not closed.

[image]

PS You chose a very strange way to construct geometry.

NathanLKing, 2024-11-25:
Yes, that's the point. Freecad won't close it when I choose to add the tangent constraint.
What's strange about it?

tarman3, 2024-11-25:
Tangent does not mean the curves are connected.

tarman3, 2024-11-25:
PS
I strongly recommend replacing bspline with a pair of arcs

NathanLKing, 2024-11-25:
> Tangent does not mean the curves are connected.
I am aware.
When I add the coincident constraint it still shows wire isn't closed.

NathanLKing, 2024-11-25:
Ah, I found the problem. 
For some reason the Trim tool wasn't taking all of the lines, trimmed some off and left the rest.

NathanLKing, 2024-11-25:
Yet, of course, there's another issue. It won't allow edits of the spline.

tarman3, 2024-11-25:
[image]

NathanLKing, 2024-11-25:
Exactly.... 
Seems yet another issue occuring when I initiate a spline and connect it to a point via a coincident constraint

tarman3, 2024-11-25:
Replace bspline with arcs and the problems will solve themselves

NathanLKing, 2024-11-25:
I'm trying that now.
Doesn't actually solve the FreeCAD errors with splines

NathanLKing, 2024-11-25:
That's a whole lot of steps to mend the errors created by FreeCAD.

tarman3, 2024-11-25:
There are many errors created by you))

NathanLKing, 2024-11-25:
I simply used the constraints that are built into FreeCAD

tarman3, 2024-11-25:
Over time, experience in optimal problem solving will come

NathanLKing, 2024-11-25:
Stop with the dribble. You're being arrogant and unhelpful.

tarman3, 2024-11-25:
I hope my video helped you.
Have a good day.

luzpaz, 2024-11-25:
OP is asking for UI to guide user to a solution. Any ideas how this can happen ?

maxwxyz, 2024-11-25:
what is the actual issue here?

luzpaz, 2024-11-25:
@NathanLKing would you mind succinctly describing what is the issue and what you expect FreeCAD to do ?

luzpaz, 2024-12-15:
there are a few potential issues in here. @AjinkyaDahale could you assist ?

AjinkyaDahale, 2024-12-15:
> there are a few potential issues in here. @AjinkyaDahale could you assist ?

Seems the OP has issue working with a combination of B-splines, tangent, and trim. The ideal solution would be to just create an end-to-end tangent in the first place (also consider angle constraints if tangent isn't right). However, if there's a specific issue in trim that needs to be addressed (it is possible the refactor introduced this).

The title of this issue doesn't describe that question though. There is no way to "constrain a B-spline on a circle".

Further discussions of using simpler tools may be valid, but should likely be held outside of this issue. The issue can stay even if the OP decides to move to different methods.

luzpaz, 2024-12-15:
Thanks. So we need a better name for this ticket and we need to split this ticket in regards to trim and simpler tools.

luzpaz, 2024-12-15:
@NathanLKing is the trim issue this ?  
* https://github.com/FreeCAD/FreeCAD/issues/18497

closed 2025-09-17 by github-actions[bot]