FreeCAD/FreeCAD #18119: Expression editor - Dual monitor problem

Pendelf, 2024-11-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When you call the Expression editor window, only part of it is visible if two monitors are connected to the computer. Main monitor on the left, second monitor on the right

[image]

https://github.com/user-attachments/assets/4f3976cf-e092-4616-98c7-b3fd851f405b

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39287 (Git) Conda
Build type: Release
Branch: main
Hash: b5a43eb2208b91feb76b2cc363e3956922c3ad75
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.51
  * fasteners 0.5.31
  * PieMenu 1.9.5
  * sheetmetal 0.5.6
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-11-25:
For some strange reason, I cannot reproduce. I have three screens, the left one is primary.

What is the purpose of this video?

What video card do you have? What driver version is installed?

[image]

xtemp09, 2024-11-25:
> The video shows that the value entry menu appears at the junction of the screens, not on the main screen

@Pendelf, are you trying to say that _ExpressionCompleter_ widget pops up on a secondary screen? This is an intended behavior. If you move the dialog slightly to the left, the completer shows up on the primary screen.

The widget is like a menu ― it shows up on a screen "completely". To do so, it determines the position to show up, on one screen or another.

Are you suggesting that the completer has to show up between the screens? I find this idea ludicrous. Screens simply can have different resolution or other settings.

Pendelf, 2024-11-25:
No on the contrary, let the menu appear on the main screen not on the intersection. It is inconvenient when you have part of the menu on one screen and the other on the other.

Pendelf, 2024-11-25:
Here's an example of how it works in LinkStage and in the main branch, in LinkStage the menu appears on the main screen, all handy

https://github.com/user-attachments/assets/f4aec8a1-9fde-49ac-b6f7-920e005b041f

xtemp09, 2024-11-27:
I have just written the code to adjust the position of expression editor, sadly, it has no effect, since _KDE_ adjusts window's position automatically if the window goes offscreen. Let's wait until _Windows_-owning developer finds time to close this issue.

ChuckInGame, 2024-12-08:
is it possible to make the expression editor a bit bigger and moveable ?
Currently, the editor is not moving, the hole freecad main window is moving:

https://github.com/user-attachments/assets/ab62fede-2e88-4ab5-9b87-7934348ec9c2

Pendelf, 2025-08-27:
@maxwxyz In the latest versions, the issue no longer occurs; it might make sense to close it

closed 2025-08-27 by maxwxyz